raspberrypi/rpi-imager #107: Unspecified libcurl error

arkabase, 2020-09-21:
Hi,
just installed rpi-imager 1.4 on windows 10
starts to write rpiOS lite on sdcard then stops with "unspecified libcurl error"
tried with different cards, same result !
have I missed something ?
Thanks
Marc

maxnet, 2020-09-21:
Not sure what the problem is.
Means the library we use to download files from the Internet (libcurl) indicated that download failed, but did not fill in any error message.

You do can download https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/2020-08-20-raspios-buster-armhf-lite.zip manually with a web browser?
Downloading something that is not from raspberrypi.org like LibreELEC does work?

AdriaanKuipers, 2020-09-29:
I just had the same problem.

Tried another time without doing or changing anything, worked fine the second time...

lurch, 2020-09-29:
Hmmm, I wonder if something like "Download failed. Please check your internet connection and try again later." would be a more user-friendly error message? :person_shrugging:

maxnet, 2020-09-29:
>Hmmm, I wonder if something like "Download failed. Please check your internet connection and try again later." would be a more user-friendly error message?

Well, I am not sure if it is the user's own Internet connection that is the problem.

If the user was able to select the image, it does mean it was able to download the OS list at first, so it is not like the computer is not connected to the Internet at all.
And if I use a Windows virtual machine, and simulate disconnecting network after that point, it displays an "Couldn't resolve downloads.raspberrypi.org" error on pressing write:

[image]

And not the "unknown libcurl error".

Wonder if the error is perhaps caused by the webserver returning an HTTP error code on download due to some temporary problem.
It could be that libcurl does not fill in an error message for that, because the error message returned by the server would be in the HTTP response body instead.
Do should be possible to test for that, and perhaps also display the server IP that gave the error to the user in that case.
(downloads.raspberrypi.org resolves to multiple different servers, so if one of them gives problems at times, it may be useful knowing which one).
But this is all just guessing.

Would like to know if anyone experiencing this error, do is able to write another image that comes from a different server (e.g. LibreELEC) at the time of error, which would support this theory.

lurch, 2020-09-29:
ping @XECDesign as he's the one that looks after downloads.raspberrypi.org so might have some ideas.

XECDesign, 2020-09-29:
Sorry, no idea. I only upload the files, everything else is done by MB.

closed 2021-01-17 by maxnet