raspberrypi/rpi-imager #108: (v1.4) Not able to access network drives

GeekyTim, 2020-09-30:
I have a network drive mounted on my PC (with drive letter) but it does not show up in the imager.  I have also tried specifying the network path, but that does not work either.

I'm using Windows 10 2004.

maxnet, 2020-09-30:
- Mapped network drives are only mapped for the logged in user. Imager runs as admin on Windows.
- Do should be possible to enter \\1.2.3.4\name-of-share in the file name field, if the password to access said share is saved on system (e.g. you accessed \\1.2.3.4\name-of-share earlier through explorer)

lurch, 2020-10-01:
Related to #65 ?

maxnet, 2020-10-02:
>Related to #65 ?

In #65 the user did was able to access the network share in the open file dialog. Just failed at that time when pressing write button.

andrum99, 2020-12-03:
> * Mapped network drives are only mapped for the logged in user. Imager runs as admin on Windows, not as the user. So it is normal for that to not work.

It can be made to work via a registry setting `EnableLinkedConnections`- see https://www.winability.com/how-to-make-elevated-programs-recognize-network-drives/. I seem to have it set on my system -  I suspect I've probably tweaked it myself and forgotted I'd done it.

tdewey-rpi, 2023-11-17:
Closing as stale. Please raise a new issue if you think this still applies.