raspberrypi/rpi-imager #110: *Suggestion* Have an option to choose which raspberry pi's that you have

jack60612, 2020-10-23:


lurch, 2020-10-23:
AFAIK, one of the driving factors for RPi-Imager was to make it as easy to use as possible. And it sounds like this suggestion would only make it more complex? :confused: 

[NOOBS](https://github.com/raspberrypi/noobs) and [PINN](https://github.com/procount/pinn) automatically know what Pi they're being run on, so they **do** do the filtering you're asking for.

jack60612, 2020-10-23:
like an advanced option or something

mobluse, 2021-01-18:
Some Pis e.g. Pi Zero, Pi 1 A, Pi 4 B can be connected to the computer that runs Imager and thus be automatically identified and also used as an SD card writer/reader.

maxnet, 2021-01-18:
>Some Pis e.g. Pi Zero, Pi 1 A, Pi 4 B can be 
>connected to the computer that runs Imager and
> thus be automatically identified and also used 
>as an SD card writer/reader.

Correct me if I am wrong.
But I don't think Pi 4 is supported by rpiboot, since on CM4 it requires setting a jumper to disable bootrom and get it to work.

And not sure if the target audience of the models that do support usb boot need help with figuring out which Image is supported by their board.

cillian64, 2023-10-19:
This is implemented by #645 which will be included in the 1.8.0 release coming shortly

closed 2023-10-19 by cillian64