raspberrypi/rpi-imager #112: Text on Ubuntu options is slightly wrong

andrum99, 2020-10-25:
There's two minor issues with the Ubuntu entry in the OS list:

1. The description for the Ubuntu entry in the list needs updated to mention the desktop variant
2. The description for the Ubuntu Desktop image says `4Gb+` at the end - this should be `4GB+`

1. is defined by https://downloads.raspberrypi.org/os_list_imagingutility.json.

2. is defined by https://changelogs.ubuntu.com/raspi/os_list_imagingutility_ubuntu.json and is Canonical's responsibility.

lurch, 2020-10-26:
ping @waveform80

waveform80, 2020-10-31:
I'll see what I can do about number 2 on that list (for reference the repo that holds our `os_list_imagingutility_ubuntu.json` file, along with the script we use to update it, can be found at https://code.launchpad.net/~ubuntu-core-dev/meta-release/ubuntu).

Not much I can do about the first item though (we did notice it prior to groovy's release, and someone was asked to coordinate with the pi foundation to get it updated but I guess that didn't happen). I'm happy to post a patch for `os_list_imagingutility.json` here if that's preferred, but that seems a bit odd given it's not in this repo. Perhaps it could be moved to a public repo (assuming it's in a repo)?

andrum99, 2020-10-31:
> I'll see what I can do about number 2 on that list (for reference the repo that holds our `os_list_imagingutility_ubuntu.json` file, along with the script we use to update it, can be found at [...]

Thanks for looking at the Ubuntu side. I suspect the Pi side is not in a source repo, as it is just a file on the main download server: perhaps @lurch could ping whoever looks after that?

lurch, 2020-11-01:
ping @XECDesign :slightly_smiling_face:

XECDesign, 2020-11-02:
Sorted. Thank you all.

closed 2020-12-07 by maxnet

andrum99, 2020-12-07:
Ubuntu side seems to still be outstanding - ping @waveform80, as it seems I can't submit a PR on Canonical's repo.