raspberrypi/rpi-imager #113: [Windows][V1.4] Program does not start properly

rohabu, 2020-10-30:
I've installed V1.4, rebooted and started the program but the main window doesn't show up.
The program is shown in the taskbar but shows only a white previe (see linked picture: https://ibb.co/k6NWd2n )
I reinstalled the application but it doesn't changed the behaviour.

Can anyone assist me what could be wrong / missing?

best regards
Rolf

maxnet, 2020-10-30:
This does is a normal computer setup which you are using locally with a monitor and keyboard?
Other applications that use graphics acceleration -such as games- do work properly on this computer?

(Someone else reported a similar issue, but he was using VNC to access his laptop remotely  through VNC because screen had broken. The VNC server used did not like accelerated applications).

rohabu, 2020-10-30:
Yes, it is a normal computer and i access it normally with a connected monitor and keyboard/mouse
Games, Adobe Premiere, Photoshop etc. work normally.
Is there some kind of log or something where i can get more informations?

maxnet, 2020-10-30:
If you open a command line prompt, and run it as:

```
cd \program files (x86)\raspberry pi imager
rpi-imager --debug
```

It may print some stuff to console.
However it is not likely to generate useful information if it does start properly (no errors, no crashes), but just graphics are not displaying properly.

rohabu, 2020-10-30:
Okay, i started the program using the --debug switch and it started flawlessly.
After this test i started it from the startmenu and it also worked.

The only difference to my other tests were, that i had connected my laptop to the dock and worked on an external monitor.
I rememberd that i installed Imager a while ago and i'm pretty sure i had it connected to my external monitor and when i now start the programm without the connection to the external monitor it wont show up on the laptop monitor.

maxnet, 2020-10-30:
Ah, thought your problem was a white application screen.
But is actually just that the window is off-screen.

Duplicate of #91 (Fixed in git)

closed 2020-10-30 by maxnet