raspberrypi/rpi-imager #118: Modify .nsis instalelr script to switch to multilanguage

bovirus, 2020-11-19:
@raspberrypi

Hello.

You are using NSIS to create the Windows installer.

Could you please add relative instructions in NSIS instalelr script to switch from mono (English) language to multi language installer?

Please add all custom strings used in the installer to enable the translator to help you to have as much as possible languages for the installer.

Thanks.

maxnet, 2020-11-19:
Think the only custom string used is the message if (an old version of) rpi-imager.exe is still running when setup.exe is executed.
"Raspberry Pi Imager seems to be running and busy.$\nDo you want to terminate process?"

Other than that it is only standard NSIS stuff.

bovirus, 2020-11-19:
1 - ENG
"Raspberry Pi Imager seems to be running and busy.$\nDo you want to terminate process?"
2 - ENG
"Windows 7 or above required"

1 - ITA
"Raspberry Pi Imager sembra essere in esecuzione e occupato.$\nVuoi terminare il processo Raspberry Pi Imager?"
2 - ITA
"Per l'esecuzione del programma è richiesto Windows 7 o versioni successive"

The string in Italian that I wrote are longer because explain more details about the context.

What kind of NSIS installer are you using?
Could be necssary if it's n old version (2.xx) an update of Italian NSIS language.

bovirus, 2020-11-19:
I found also the link in program group about rpi-image website like

"$INSTDIR\${APP_NAME} website.url"
"$SMPROGRAMS\$SM_Folder\${APP_NAME} Website.lnk" "$INSTDIR\${APP_NAME} website.url"

Delete "$INSTDIR\${APP_NAME} website.url"
Delete "$SMPROGRAMS\$SM_Folder\${APP_NAME} Website.lnk"

Delete "$SMPROGRAMS\Raspberry Pi\${APP_NAME} Website.lnk"

Could be replace by two variables where are translated in different langauge

"$INSTDIR\${LSTR_01}.url"
"$SMPROGRAMS\$SM_Folder\${LSTR_01}.lnk" "$INSTDIR\${LSTR_01}url"

Delete "$INSTDIR\${LSTR_01}.url"
Delete "$SMPROGRAMS\$SM_Folder\${LSTR_01}.lnk"

Delete "$SMPROGRAMS\Raspberry Pi\${LSTR_01}.lnk"

where 

English string
LSTR_01="${APP_NAME} website"

Italian string
LSTR_01="Sito web ${APP_NAME}"

maxnet, 2020-11-19:
>What kind of NSIS installer are you using?

3.0.5
But it don't know if that is multi-language by default, or if one needs to add entries to script to load languages or something.
Never used it for other languages.

>I found also the link in program group about rpi-image website like

I am not a big fan of changing file names and short cuts depending on language.
That complicates things if a user already has Imager installed, and want to install a newer version overwriting files.

Then the user will have both a "Raspberry Pi Imager website.lnk" and a "Sito web Raspberry Pi Imager.lnk"
And on uninstallation only the latest "Sito web" one will be removed.

Unless you also add logic to the nsis script to call uninstaller first on upgrading existing installation.

bovirus, 2020-11-19:
@maxnet 

NSIS 3.05 has the Italian langauge updated.

I already get this isssues in teh apst and otehr developers sugegst to do that.

About "xxxx.lnk" or "xxx.url" when youw ant to remove you can simply use use command string like "*.url" or "*.lnk" and you will delete all .lnk or .url without problem about which language was set before and after a new installation.

Example instead

Delete "$SMPROGRAMS\Raspberry Pi${APP_NAME} Website.lnk"

use

Delete "$SMPROGRAMS\Raspberry Pi*.lnk"

maxnet, 2020-11-19:
What if Raspberry Pi wanted to release other Windows software in the future?
User only wants to uninstall Imager, but it may delete the other program as well, if it matches "Raspberry Pi*.lnk" ...

bovirus, 2020-11-19:
Raspberry Pi is the program group then when you will use uninstall you will delete ONLY the lnk in Raspberry PI program group (not all links).

maxnet, 2020-11-19:
>Raspberry Pi is the program group then when you will use uninstall you will delete ONLY the 
>lnk in Raspberry PI program group (not all links).

But Imager could theoretically share the program group "Raspberry Pi" with other software made by "Raspberry Pi"

bovirus, 2020-11-19:
Youi can create s new program group under"raspberry PI" for each tool.
Then you have  Raspberry Pi -> rpi-image for rpi.imager tool.
I believe that installer can be translated in a second step.

For me most important things is to add new GUI translations (total 8 languages).
If you can for me it's more flexible to have external language files in translations subfolder as .ts/.qm language files.
Enable to check/change/test translation in place (ex. for Windows platform).

maxnet, 2020-11-19:
>To avoid this problem you can create s new program group under "Raspberry PI" for each tool.

But then the user will also see multiple levels in the "start" menu in Windows, and that may not be desired.

Anyway, I just remembered we are currently not even creating a shortcut for the website.
That part of the standard NSIS script used is only used "!ifdef WEB_SITE" and we are not setting WEB_SITE.
So the website link we do not have, does not need to be translated :-)

bovirus, 2020-11-19:
Infact I didn't find in current program group in version 1.4.

Then we can apply the changes to translation languages.

commit a37f9a9f (2020-11-24): Windows: NSIS installer: enable multi-language (en/it/nl)

Closes #118
closed 2020-11-24 by maxnet