raspberrypi/rpi-imager #129: EEPROM image unable to write to SD card

joebusify, 2020-12-03:
When using the EEPROM image option, it does not ask for the user credentials, and thus is unable to format the drive and write the image. (RPI Imager on Ubuntu 20.10)

maxnet, 2020-12-03:
Do you get any error message?

The "erase, format as FAT32" option does not work for you either?

Note that it is normal that no authorization is asked, as it are not privileged operations under Ubuntu.
Only raw disk access is (which is not needed for the EEPROM files).

maxnet, 2020-12-03:
BTW you did install the Ubuntu .deb from: https://www.raspberrypi.org/software/ ?

(As opposed to the snap from Canonical)

maxnet, 2020-12-05:
Probably related to: https://github.com/popey/imager-snap/issues/8

Not reproducible with our own .deb.

closed 2020-12-05 by maxnet