raspberrypi/rpi-imager #130: RPI Imager Will Not Format 64GB USB Drive to FAT32

eeyatahn, 2020-12-06:
I tried to use Raspberry Pi Imager v1.4 to format a large USB drive (64GB) to Fat32. I selected "Erase format to fat32" and my USB drive. But RPI Imager still formated it to exFAT.

It was my understanding that RPI Imager would change the format to FAT32 from an exFAT formated USB drive. It does not seem to do that.

maxnet, 2020-12-06:
Which operating system are you using?

Imager's workflow differs by platform.
On Linux and Mac we outsource formatting to the system.
For Windows we use fat32format.exe that we bundle with Imager.
fat32format does not know the exFAT format at all. So if you end up with an exFAT on Windows it is more likely that the card already was exFAT formatted before it started, and it failed to write to the card.

eeyatahn, 2020-12-07:
Windows 10 20.04

Yes the USB drive was already formated in exFAT 
I was trying setup a PINN system for the PI but needed to have the USB drive in FAT32 format. I could have done it in windows but that would mean a full format of the drive. I finally wound up formating the drive in Ubuntu using Gparted. 

Is there a way to force RPI imager to override the format if the drive is already formated in exFAT?

maxnet, 2020-12-07:
>I was trying setup a PINN system for the PI but needed to have the USB drive in FAT32 format.

Btw is it a Pi 4 or an older model?

As some older generation Pi have problems with large FAT32 partitions as well.
(Or to be more exact the bootrom had some bug in which it does not like it if the files to be booted are located on sd card above a certain offset. Which in practice can only occur if fat partition is large. But even if partition is large, it may not always occur, as it depends where os stored the files within partition.)

>Is there a way to force RPI imager to override the format if the drive is already formated in exFAT?

It is supposed to do that already with the ”format as fat32” option.
Only mentioned that if it failed to do its job, things still may be exfat, but that is not how it is supposed to be.

If you suspect that it failed to write to sd card, you can run fat32format from command prompt manually and see if it spits out an error message which may explain what is going on.
Among the lines of:

‘’’
cd \program files (x86)\raspberry pi imager
fat32format.exe x:
‘’’

Where x: is your sd card.

Btw how did you check what file system type it was exactly?

eeyatahn, 2020-12-07:
> Btw is it a Pi 4 or an older model?

New Raspberry Pi 4 Model B+ 4 GB. Only about a month old.

> If you suspect that it failed to write to sd card

It was not an SD card as I mentioned it was a Memorx 64GB USB ThumbDrive. 

Though I have had this same problem using a San Disk Ultra 64 GB SD card as well when trying to install PINN. The PI would not Post, Boot or even turn on HDMI. 
I had to insert, into the "Target" field of the shortcut, with the help of Procount.

> --repo https://raw.githubusercontent.com/procount/pinn-os/master/iu/os_list_iu_pinn.json 

 To force RPI-Imager to install PINN. 

> Btw how did you check what file system type it was exactly?

Yes, Windows Disk Managment reported it as exFAT and Ubuntu running Gparted also reported it as exFAT. It was with Gparted that I was able to format the USB ThumbDrive with FAT32 because RPI-Imager would not.

eeyatahn, 2020-12-07:
Well I am confused.... Just tired RPI-Imager.... Formated the USB ThumbDrive as exFAT... Then loaded RPI-Imager.... and....

It changed the format to FAT32.

I have no idea why if would not before. But it does seem to be working now....

Thank you for your help.

eeyatahn, 2020-12-07:
Update.....

Well everything seems to work now... Sorry for the confusion I will chock it up to a Windows / Microsoft Brain Fart.... I really don't know why.

Sorry to have bothered you Thank you

maxnet, 2020-12-07:
Good to hear it is solved.

closed 2020-12-07 by maxnet