raspberrypi/rpi-imager #132: [Feature Suggestion] Progress on Windows Task Bar

digocesar, 2020-12-07:
Would be so good when writing a SD card to see the progress on Windows Task Bar. It's common to do other things while writing a SD card but without the task bar progress we need to open the rpi-imager every time to see if it already has finished.
Looking on internet it seems that is possible:
https://www.google.com/search?q=qt+windows+taskbar+progress+bar

closed 2020-12-09 by maxnet

MagicAndre1981, 2024-12-19:
I don't see the progress in Windows 10 with StartIsBack++

[image]

Do I need to enable this? @maxnet

lurch, 2024-12-19:
Looks like the feature was removed in https://github.com/raspberrypi/rpi-imager/commit/103924eff0b234e5437e8842a27eba42a47fde64

MagicAndre1981, 2024-12-20:
ok, thanks.

lurch, 2024-12-20:
I obviously can't speak directly for @tdewey-rpi (the maintainer of this repo), but it seems unlikely that we'd start using 3rd-party forks of (parts of) Qt.

tdewey-rpi, 2024-12-20:
@lurch has accurately captured my position. Essentially, using that module introduces a friction point for moving to future Qt releases that is undesirable.

I may decide to implement such functionality directly in Imager, however I very much consider this a "nice-to-have", rather than a hard requirement.

commit 1d008792 (2020-12-09): Windows: show progress in taskbar

Windows: show progress in taskbar
Closes #132

Implemented in ImageWriter class instead of qml for practical
reasons. GUI stuff does not really belong there, but there is no
easy way to have platform specific stuff in qml, lacking #ifdef