raspberrypi/rpi-imager #133: Windows task bar progress writing Custom Image

digocesar, 2020-12-18:
Task #132 added the progress status on Windows Task Bar. But when writing a Custom Image it is always as "indeterminate" status.

lurch, 2020-12-18:
What file-format is your Custom Image in?

maxnet, 2020-12-18:
And is it specific to the Windows task bar?
As in the main window does show correct progress?

digocesar, 2020-12-18:
I'm using a .img file.
The issue is only with status progress on Windows Task Bar. The progress bar on applications is correct.

digocesar, 2020-12-18:
See the error:
https://user-images.githubusercontent.com/17618149/102651212-eb89e400-414a-11eb-9d63-182baf06c2c5.mp4

maxnet, 2021-01-17:
>See the error:
>https://user-images.githubusercontent.com/17618149/102651212-eb89e400-414a-11eb-9d63-182baf06c2c5.mp4

Not reproducible to me:

[image]

So problem may be Windows version or timing specific.

digocesar, 2021-01-29:
Here is the debug output:
```
C:\Program Files (x86)\Raspberry Pi Imager>DirectWrite: CreateFontFaceFromHDC() failed (Indica um erro em um arquivo de entrada, como um arquivo de fonte.) for QFontDef(Family="MS Shell Dlg 2", pointsize=8.25, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Roboto", lfWidth=0, lfHeight=-11) dpi=96
DirectWrite: CreateFontFaceFromHDC() failed (Indica um erro em um arquivo de entrada, como um arquivo de fonte.) for QFontDef(Family="MS Shell Dlg 2", pointsize=8.25, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Noto", lfWidth=0, lfHeight=-11) dpi=96
setGeometry: Unable to set geometry 680x420-1199+310 (frame: 702x476-1210+265) on ApplicationWindow_QMLTYPE_3_QML_24/"" on "\\.\DISPLAY2". Resulting geometry: 1022x633-1196+324 (frame: 1044x689-1207+279) margins: 11, 45, 11, 11 minimum size: 680x420 maximum size: 680x420 MINMAXINFO maxSize=0,0 maxpos=0,0 mintrack=702,476 maxtrack=702,476)
setGeometry: Unable to set geometry 1022x633-1196+299 (frame: 1044x689-1207+254) on ApplicationWindow_QMLTYPE_3_QML_24/"" on "\\.\DISPLAY2". Resulting geometry: 680x420-1196+299 (frame: 702x476-1207+254) margins: 11, 45, 11, 11 minimum size: 680x420 maximum size: 680x420 MINMAXINFO maxSize=0,0 maxpos=0,0 mintrack=702,476 maxtrack=702,476)
QObject::setParent: Cannot set parent, new parent is in a different thread
[INFO] Enumerating USBSTOR
[INFO] Enumerating SCSI
[INFO] Enumerating SCSI
[INFO] Enumerating USBSTOR
[INFO] Enumerating SCSI
[INFO] Enumerating SCSI
[INFO] Enumerating USBSTOR
[INFO] Enumerating SCSI
[INFO] Enumerating SCSI
device "\\\\.\\PhysicalDrive2"
Removing partition table from Windows drive # "2" ( "\\\\.\\PhysicalDrive2" )
[INFO] Enumerating USBSTOR
[INFO] Enumerating SCSI
[INFO] Enumerating SCSI
Locking volume "\\\\.\\E:"
Locked volume
Hash of uncompressed image: "28c6de18f67df0f3f4fb6c1f1a70d4c506e603ce3c2277bf5b8dd93fec79f0f8"
Write done in 677 seconds
Verify done in 989.61 seconds
Verify hash: "28c6de18f67df0f3f4fb6c1f1a70d4c506e603ce3c2277bf5b8dd93fec79f0f8"
Stopping storage services
O serviço de Serviço de Armazenamento está sendo finalizado .
O serviço de Serviço de Armazenamento foi finalizado com êxito.

Writing first block (which we skipped at first)
Unlocked volume
mountutils: \\.\PhysicalDrive2
mountutils: Checking whether drive is ejectable: C
mountutils: The drive is fixed
mountutils: Opening drive letter handle: C
mountutils: Closing drive letter handle
mountutils: Continuing with the next available letter
mountutils: Checking whether drive is ejectable: D
mountutils: The drive is fixed
mountutils: Opening drive letter handle: D
mountutils: Closing drive letter handle
mountutils: Continuing with the next available letter
mountutils: Checking whether drive is ejectable: E
mountutils: The drive is removable
mountutils: Opening drive letter handle: E
mountutils: Closing drive letter handle
mountutils: Drive letter device matches
mountutils: Ejecting drive letter
mountutils: Creating volume handle
mountutils: Volume is not mounted
mountutils: Retrying
mountutils: Ejecting drive letter
mountutils: Creating volume handle
mountutils: Locking volume
mountutils: Dismounting volume
mountutils: Ejecting volume
mountutils: Volume ejected
mountutils: Unlocking volume
mountutils: Closing volume handle
mountutils: Retrying
mountutils: Ejecting drive letter
mountutils: Creating volume handle
mountutils: Couldn't create volume handle
mountutils: Drive letter has already been ejected
mountutils: Continuing with the next available letter
QObject: Cannot create children for a parent that is in a different thread.
(Parent is LocalFileExtractThread(0x5939a48), parent's thread is QThread(0x9105a8), current thread is LocalFileExtractThread(0x5939a48)
Restarting storage services
O serviço de Serviço de Armazenamento está sendo iniciado.
O serviço de Serviço de Armazenamento foi iniciado com êxito.
```

Is the error `QObject::setParent: Cannot set parent, new parent is in a different thread` related to this problem?
Can anyone generate a release with more debug info to understand what is wrong?

maxnet, 2021-01-29:
>Is the error QObject::setParent: Cannot set parent, new parent is in a different thread related to this problem?

No, that is "normal"

maxnet, 2021-01-30:
Managed to reproduce your problem.
Only happened on images >2GB and I usually test with small stuff to save time.

closed 2021-01-30 by maxnet

digocesar, 2021-02-01:
Great! Thanks!

commit b72f3a61 (2021-01-30): Fix Windows progress on images >2GB

Closes #133
commit 2261faab (2021-01-30): Fix Windows progress on images >2GB

Closes #133