raspberrypi/rpi-imager #136: Enable GitHub Discussions for this project?

geerlingguy, 2020-12-23:
I think, judging by the fact that there are continuous Issues submitted for adding some sort of 'enable SSH' functionality, it might be nice to have Discussions enabled on this GH repository, and we could have a discussion (with input from the community of users, maintainers, and RPi representatives) around some of the reasons why or why not to include a feature like that, how it could be implemented so it's not a footgun, etc.

That seems like a better plan than just having someone create a new issue for it every month or so, then closing it :)

lurch, 2020-12-23:
Just to play devil's advocate: are the people who don't find any of the closed enable-SSH-request issues before posting their own, any more likely to find a "discussion" about why enable-SSH hasn't been added before posting a new issue requesting it? :shrug: 

(disclaimer: I dunno how widely-used "GitHub Discussions" are)

ghollingworth, 2020-12-23:
I would agree there really isn't any point here.

My reasons are legion, mostly around making it simple (and secure) for beginners, but also making sure it works across all operating systems (i.e. adding a file to /boot will not work for LibreELEC)

This software is specifically for beginners, if you want something more complex, look at PiBakery (which is what this is based on anyway!)

https://github.com/davidferguson/pibakery/releases/tag/v2.0.0

lurch, 2020-12-23:
...but note that PiBakery seems to have stagnated https://github.com/davidferguson/pibakery/issues/180

geerlingguy, 2020-12-23:
@ghollingworth - I was going to give a few suggestions as to how it could maybe be done in a way that's not a footgun, but that was a secondary point—I find it's nice to have a 'Discussions' place for things where the maintainers may not want it to be clogging up the issue queue.

It's a _very_ new feature (just out of private beta a couple weeks ago), but I like the fact that discussions don't have a count at the top of the tab, and are a bit more informal.

geerlingguy, 2020-12-29:
@ghollingworth - As an alternative, if you could add a preference option for 'don't unmount after writing the image', that would also be helpful in making it so users who do want to set up WiFi, SSH, etc. wouldn't have to physically re-plug the card to get it to mount again, or manually re-mount it.

lurch, 2020-12-29:
> if you could add a preference option for...

I wasn't involved in the development of rpi-imager, but I _suspect_ that it was a deliberate decision to not have **any** preference options in Raspberry Pi Imager ?

ghollingworth, 2020-12-30:
Indeed,

The whole idea behind this application is to make it foolproof...  I'm not saying there's a lot of fools out there, but in general, anyone who's commenting on this GitHub is not the kind of person that this application is geared towards.

I realise it's a really useful application and works really well and really easily, but adding options like this just increases the opportunity for people to do something wrong when they image their SD cards.  Here's a bunch of possible outcomes:

1) Linux or Mac don't flush writes to removable disks automatically and therefore it's possible using this option you won't see your last write (which happens to be the first MB of the card) if you remove the card too quickly.
2) Creating the ssh file in /boot doesn't work on Ubuntu, LibreELEC, RISC OS...  So people will get confused...
3) Enabling SSH by default is insecure and beginners may not understand that.  It's then possible hacked IoT devices in your house will try to access your Pi using the default username and password (this is actually a thing, you can check online and see that many auto password guessers try pi, raspberry to log into things.

I agree it would be very useful for a large selection of people, but that's not who this application is designed to help.

Sorry :)

closed 2020-12-30 by ghollingworth