raspberrypi/rpi-imager #137: Compilation fails on FreeBSD, lsblk dependency unmet

probonopd, 2020-12-24:
Compilation fails on FreeBSD, lsblk dependency unmet:

```
FreeBSD:/tmp% cd rpi-imager 
FreeBSD:/tmp/rpi-imager% mkdir build
FreeBSD:/tmp/rpi-imager% cd build
FreeBSD:/tmp/rpi-imager/build% cmake ..
-- The CXX compiler identification is Clang 8.0.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
udisks2 support enabled
-- Performing Test atomicbuiltin
-- Performing Test atomicbuiltin - Success
-- Found CURL: /usr/local/lib/libcurl.so (found suitable version "7.72.0", minimum required is "7.32.0")  
-- Found LibArchive: /usr/local/lib/libarchive.so (found suitable version "3.4.3", minimum required is "3.2.0") 
-- Found OpenSSL: /usr/lib/libcrypto.so (found version "1.1.1d")  
CMake Error at CMakeLists.txt:230 (message):
  Unable to locate lsblk (used for disk enumeration)

-- Configuring incomplete, errors occurred!
See also "/tmp/rpi-imager/build/CMakeFiles/CMakeOutput.log".
FreeBSD:/tmp/rpi-imager/build% lsblk
zsh: command not found: lsblk
```

Perhaps `lsblk` needs to be replaced with something else for FreeBSD:
https://forums.freebsd.org/threads/list-block-devices-on-freebsd-lsblk-8-style.72447/

maxnet, 2020-12-24:
>Perhaps lsblk needs to be replaced with something else for FreeBSD

Or perhaps enumerate storage using udisks2 interface.
No idea if the BSD implementation ( https://foss.heptapod.net/bsdutils/bsdisks ) is feature complete enough though.

hbarnard, 2022-04-03:
I expect this is something simple that I don't fully understand. However, this is the failure from my attempted compile this morning. I'm trying a recompile because I'm getting verification failures on:

`Linux version 5.0.0-32-generic (buildd@lgw01-amd64-015) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)) #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019
`
sometimes when the image succeeds it doesn't boot too.

```
In function ‘int main(int, char**)’:
/home/hbarnard/projects/rpi-imager/src/main.cpp:273:19: error: ‘class QApplication’ has no member named ‘screenAt’; did you mean ‘screens’?
         if ( !app.screenAt(QPoint(x,y)) || !app.screenAt(QPoint(x+w,y+h)) )
                   ^~~~~~~~
                   screens
/home/hbarnard/projects/rpi-imager/src/main.cpp:273:49: error: ‘class QApplication’ has no member named ‘screenAt’; did you mean ‘screens’?
         if ( !app.screenAt(QPoint(x,y)) || !app.screenAt(QPoint(x+w,y+h)) )
                                                 ^~~~~~~~
                                                 screens
CMakeFiles/rpi-imager.dir/build.make:113: recipe for target 'CMakeFiles/rpi-imager.dir/main.cpp.o' failed
make[3]: *** [CMakeFiles/rpi-imager.dir/main.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/home/hbarnard/projects/rpi-imager/src/imagewriter.cpp: In member function ‘void ImageWriter::replaceTranslator(QTranslator*)’:
/home/hbarnard/projects/rpi-imager/src/imagewriter.cpp:1179:18: error: ‘class QQmlApplicationEngine’ has no member named ‘retranslate’
         _engine->retranslate();
                  ^~~~~~~~~~~
CMakeFiles/rpi-imager.dir/build.make:137: recipe for target 'CMakeFiles/rpi-imager.dir/imagewriter.cpp.o' failed

```

lurch, 2022-04-03:
> Linux version 5.0.0-32-generic (buildd@lgw01-amd64-015) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)) #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019

@hbarnard It's not possible to compile RPi Imager on Ubuntu 18.04, because RPi Imager uses a newer version of QT than is included in that old version of Ubuntu. See #197 

> I'm trying a recompile because I'm getting verification failures

I'm afraid you're barking up completely the wrong tree. A recompilation would _only_ be needed if RPi Imager doesn't run at all - if it does run, then you don't need to recompile it.
Verification errors (and non-booting images) sounds more like a problem with your SD-card and/or your SD card-reader.

(also this issue #137 is about compilation on FreeBSD, so doesn't seem like an appropriate place for comments about compilation on Ubuntu Linux??)

Oh, and if you're using the snap-installed version of RPi Imager (e.g. from the Ubuntu Software store) then try uninstalling that and use the `rpi-imager_1.6.1_amd64.zip` from [here](https://github.com/raspberrypi/rpi-imager/issues/197#issuecomment-811224673) instead. (the snap version of RPi-Imager is maintained by a 3rd-party, and seems to have problems sometimes, e.g. #369 )

hbarnard, 2022-04-03:
Thanks. I've gone back to preparing the cards on a Windows system for the moment.

---------
https://www.hughbarnard.org
Twitter: @hughbarnard

tdewey-rpi, 2024-02-06:
Closing as stale. OP's last comment was they had changed platform, rendering the report moot.

closed 2024-02-06 by tdewey-rpi

hbarnard, 2024-02-06:
Fine. Working on ubuntu now, 'anyway'.

probonopd, 2024-02-10:
> Closing as stale. OP's last comment was they had changed platform

Not sure I understand this. I am still running FreeBSD and I am still unable to run the RPi Imager.

maxnet, 2024-02-10:
>Closing as stale. OP's last comment

Note that there are two persons posting in this thread.
The OP that would like to use Imager on FreeBSD, and still does not have a solution for that.

And the last poster that was unlucky compiling on a 2018 Ubuntu version.

Entirely different problems.
Think there do is already a different issue for FreeBSD though.
Also has a Qt qml problem, in addition to the disk enumeration one.

hbarnard, 2024-02-10:
The imager is fine on the newer Ubuntu releases, so my original problem has evaporated. Just the FreeBSD now.

---------
https://www.hughbarnard.org

***@***.***
Mastodon: @***@***.***
Book: (awaiting print run) pdf/epub: https://tinyurl.com/2s4hm33b

tdewey-rpi, 2024-02-12:
Thanks all for the additional comments.

I'm still, however, not going to reopen this issue. Raspberry Pi Imager is packaged for Windows, Debian, Raspberry Pi OS and macOS as these platforms are the most likely to have new and inexperienced users coming to the Raspberry Pi ecosystem.

The argument against providing explicit support for FreeBSD is much the same as AppImages, RPMs, other OS' and packaging schemes:  there isn't the bandwidth to support them without compromising the support offered to a larger proportion of the community.

As such, issues raised against rpi-imager for building on FreeBSD will not be addressed by myself, and in the absence of a supporting PR I will be closing them as 'Won't Fix'.

I'm happy to review a PR that widens the support net - however the submitter would have to assume maintenance responsibility for the reason specified above, and I would expect them to be responsive to issues their patch may cause on our existing supported platforms.

And as ever, if the observations that support this argument turn out to be incorrect, I will happily change my position.

For my future self: Adjusting close from 'Stale' to 'Won't fix'.

emaste, 2026-08-18:
Pull request open for FreeBSD support: https://github.com/raspberrypi/rpi-imager/pull/1655