raspberrypi/rpi-imager #138: Unable to open dmg image on mac

lidderupk, 2020-12-24:
Downloaded imager-1.5.dmg on Mac version 10.15.6. I get an error `No mountable file systems". 

[image]

maxnet, 2020-12-24:
Tried downloading it a second time?

lidderupk, 2020-12-24:
I did try a bunch of times. From raspberry site. and from github. Did it work for you? I am going to restart my mac. Saw that as one of the answers. Thank you.

maxnet, 2020-12-24:
>Did it work for you?

Yes, the .dmg opens fine for me.

Also may want to double-check the shasum to make sure there is nothing wrong with download.
(The mirror I got my file from, may be a different one than yours).

[image]

maxnet, 2020-12-24:
BTW you do can successfully open .dmg files from other vendors?

And don't have anything special installed that installs kernel drivers that can affect file system operations such as USB drive encryption software and non-Apple filesystem drivers?

lidderupk, 2020-12-25:
Restarting worked. Not sure what was going on. Thank you for your help.

maxnet, 2020-12-25:
Good to hear it is solved.

closed 2020-12-25 by maxnet