raspberrypi/rpi-imager #141: RPi Imager crashes Windows 10 after verifying

diatomicDisaster, 2021-01-02:
I have experienced a consistent bug in which RPi Imager crashes Windows 10 after writing the 64-bit Raspbian disk image to an SD card. Windows blue screens and the only information given is the error KERNEL_MODE_HEAP_CORRUPTION

maxnet, 2021-01-02:
While I certainly will not pretend that Imager is free of bugs, kernel mode blue screens are typically a sign something is wrong with a driver, and not the application (which lives in user mode instead).

Previously people have reported similar problems with Dell laptops that come with an internal Realtek SD card reader that uses custom drivers.
(Most other SD card readers do not use custom drivers, but use the generic USB mass storage drivers written by Microsoft instead)
See: https://github.com/raspberrypi/rpi-imager/issues/94

Do you happen to have such SD card reader?

diatomicDisaster, 2021-01-02:
Indeed I do, so that sounds as though it's a likely cause.

maxnet, 2021-01-02:
>Indeed I do, so that sounds as though it's a likely cause.

Perhaps it should be reported to Dell/Realtek.
But reaching the right persons there may be problematic.

lurch, 2021-01-02:
@diatomicDisaster According to the comments in #94 using a USB SD-card reader should be a workaround.

If you have time to do further testing, it'd be interesting to hear if Win32DiskImager and Etcher also cause the same BSOD errors when writing the same image using the internal card-reader?

diatomicDisaster, 2021-01-02:
I used BalenaEtcher in the end which does not have the same problem. Though it's also worth noting that RPi imager still writes the image correctly and it boots fine, it just causes Windows to restart.

theonemcdonald, 2021-02-01:
https://github.com/raspberrypi/rpi-imager/issues/94#issuecomment-770576053

sprotax, 2021-02-21:
I've found that changing the removal policy in the SD card driver properties from "quick removal" to "better performance (default)" has some success in fixing the issue of Dell's crashing while flashing an image. It still crashes sometimes, I belief this is related to the state of drive when starting.
The best fix is still use a usb adaptor instead.

tdewey-rpi, 2024-02-06:
Closing as:

1. Stale
2. Not an issue with rpi-imager

closed 2024-02-06 by tdewey-rpi