raspberrypi/rpi-imager #142: Raspberry Pi Imager crashes when clicking

davidplante, 2021-01-05:
Preconditions 
1. Windows 10 Educations freshly installed
2. Raspberry Pi Imager 1.5 installed

Actions
1. Start executing rpi-imager.exe
2. Click CHOOSE OS button
3. Select Raspberry PI OS (32 bits)
4. Click CHOOSE SD CARD button

Actual result
1. SD card popups shown briefly and application crashes.

Expected result
1. Thumbdrive selectable to write image onto.

Event Viewer detail :
`Faulting application name: rpi-imager.exe, version: 1.5.0.0, time stamp: 0x00000000
Faulting module name: libstdc++-6.dll, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0x40000015
Fault offset: 0x0001f9c3
Faulting process id: 0x2d04
Faulting application start time: 0x01d6e312e3c60d41
Faulting application path: C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe
Faulting module path: C:\Program Files (x86)\Raspberry Pi Imager\libstdc++-6.dll
Report Id: 9ce805a2-8e96-4495-a66f-091466f92d89
Faulting package full name: 
Faulting package-relative application ID: 

------------------
System Information
------------------
      Time of this report: 1/4/2021, 22:36:10
             Machine name:  
               Machine Id:  
         Operating System: Windows 10 Education 64-bit (10.0, Build 19042) (19041.vb_release.191206-1406)
                 Language: French (Regional Setting: French)
      System Manufacturer: Micro-Star International Co., Ltd.
             System Model: MS-7C37
                     BIOS: Default System BIOS (type: BIOS)
                Processor: AMD Ryzen 5 3400G with Radeon Vega Graphics     (8 CPUs), ~3.7GHz
                   Memory: 32768MB RAM
      Available OS Memory: 30642MB RAM
                Page File: 8089MB used, 27416MB available
              Windows Dir: C:\Windows
          DirectX Version: DirectX 12
      DX Setup Parameters: Not found
         User DPI Setting: 96 DPI (100 percent)
       System DPI Setting: 96 DPI (100 percent)
          DWM DPI Scaling: Disabled
                 Miracast: Available, with HDCP
Microsoft Graphics Hybrid: Not Supported
 DirectX Database Version: 1.0.8
           DxDiag Version: 10.00.19041.0546 64bit Unicode
`

maxnet, 2021-01-05:
Does it also happen if there is no thumbdrive (nor other external drives) connected, and you press the "choose SD" button?

Wondering if it crashes while enumerating that drive, or some system drive.

davidplante, 2021-01-05:
Hi @maxnet. I got the very same result If I disconnect my drives.
For your information, when I start Etcher, app doesn't open and it creates 3 processes. Maybe both problems are related

maxnet, 2021-01-05:
>For your information, when I start Etcher, app doesn't open and it creates 3 processes. 
>Maybe both problems are related

Very likely.
We borrowed Etcher's drivelist library for enumerating disks...
So a problem in that will affect both projects.

The code in question ( https://github.com/raspberrypi/rpi-imager/blob/qml/dependencies/drivelist/src/windows/list.cpp ) has different methods to get the information of different drive types.
If there is a problem in a drive type that is not so common, the problem may not be reproducible to others, while it does occur to you.

Do you have any drives on your system that are of a special type?
Think of virtual drive types (generated by programs like daemon tools), iSCSI, NVMe.
Does not have to be removable drives, as the code queries all drives on the system. Even drives we end up not offering to the user to select.

==

>Faulting module name: libstdc++-6.dll, version: 0.0.0.0, time stamp: 0x00000000
>Fault offset: 0x0001f9c3

If I look at what is at that address:

```
Microsoft (R) Windows Debugger Version 10.0.18362.1 AMD64 
Copyright (c) Microsoft Corporation. All rights reserved. 

CommandLine: "C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe" 
[...]
ModLoad: 00000000`6fe40000 00000000`6ffbf000 C:\Program Files (x86)\Raspberry Pi Imager\libstdc++-6.dll 
[...]
0:000:x86> ln 6fe5f9c3 
[...]
(6fe5f860) libstdc___6!ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x163 | (6fe5fa60) libstdc___6!ZN9__gnu_cxx29__concurrence_broadcast_errorD0Ev 
```

Then I think that is a routine in libstdc++ that terminates the program if a thrown C++ exception is not caught.
Which is good in the sense that if we simply try/catch it, we should be able to at least shove the problem under the carpet so that it will no longer terminate the program.

Although finding what is causing it, will still be better.
So will see if I can add some debugging stuff to narrow that down.

lurch, 2021-01-05:
> For your information, when I start Etcher, app doesn't open and it creates 3 processes. Maybe both problems are related

If you haven't already done so, maybe it's worth creating an issue on the Etcher GH repo too? Although maybe what you're seeing with Etcher is a duplicate of https://github.com/balena-io/etcher/issues/1866 or https://github.com/balena-io/etcher/issues/3088

maxnet, 2021-01-06:
Here is a build that should catch any exception during enumerating a drive and skip that drive it is having problems with:

[imager-catch-exception.zip](https://github.com/raspberrypi/rpi-imager/files/5773264/imager-catch-exception.zip)

Can you test if it indeed no longer terminates?
And if so, start Imager in debug mode by opening a command prompt, and entering:

```
cd \program files (x86)\raspberry pi imager
rpi-imager --debug
```

It should print out some information about what exception it caught while enumerating what drive.

davidplante, 2021-01-06:
Wow, thanks guys for your help, very apreciated. I was about to install another Windows 10 version. But what I did was better.
1. Uninstall RPI-Imager
2. Unzip the archive.
3. Install RPI-Imager.
4. Running the app in Debug as suggested.

Now, here is what happened. I started the app, selected the image. Then, clicked to choose the drive. It proposed the correct drive. While writing, at the beggining, it crashed.

I've noticed it affected my partitions for the drive itself. Here are how my thumbdrive appeared before running app in debug  ([drive after](https://imgur.com/AbNkxt2)).
[image]
and after the 1st try, where writing seems to be good
[image]
I provide also the result of the debug. 
[debug.txt](https://github.com/raspberrypi/rpi-imager/files/5773371/debug.txt)

@lurch I think it would be a duplicate of 3088. Would be very interresting to check with the guys that have the Etcher issue if they have a thumbdrive showing a boot partition.

Nevertheless, I still don't know how your new installation resolved the problem.

maxnet, 2021-01-06:
>debug.txt

Thanks.
Figured out where the "basic_string::_M_construct null not valid" is coming from, and created an upstream issue: https://github.com/balena-io-modules/drivelist/issues/381

==

>Now, here is what happened. I started the app, selected the image. Then, clicked to choose 
>the drive. It proposed the correct drive. While writing, at the beggining, it crashed.

Hmm, no idea what is up with that :-(
Any message during crash?

davidplante, 2021-01-06:
No message shown. No Event created either. On the 2nd try, it works fine. Still writing the image. About 1 hour to write 88% at the moment. Never thought it would be that long.
After this, I will try Etcher again.

maxnet, 2021-01-06:
Here is a version that removes the debug stuff again.

[imager-fix-zerolength-wstr.zip](https://github.com/raspberrypi/rpi-imager/files/5775447/imager-fix-zerolength-wstr.zip)

Can you confirm that it still no longer crashes when selecting SD card?

davidplante, 2021-01-07:
Sorry for not answering earlier; my micro SD card is somehow 'broken'. Instead of showing 64GB capacity, it shows 2TB. I still can write files, but I would prefer to have a 100% working card before. It's a good quality micro SD. I'm looking for a fix for this. I can't go buy another one (full lockdown in my area).

shin-san, 2021-01-07:
Hi,

I thought I should interject as I was also following this thread. I tested the `imager-fix-zerolength-wstr.zip` version and the `Choose SD Card` has started working for me (had the same issue as david).

I also flashed an SSD in this instance and it worked as intended.

I'm attaching the debug log in case it is needed:
[debug.txt](https://github.com/raspberrypi/rpi-imager/files/5781393/debug.txt)

I'm running Windows 10 x86 Home.

maxnet, 2021-01-07:
Good to hear the patch solves it, then I'll close this issue.

Cannot help with SD cards reporting wrong capacity.
"Good quality SD card" does not say much.
Keep in mind that a high number ([some even say 30%](https://theonlinephotographer.typepad.com/the_online_photographer/2011/05/counterfeit-memory.html)) of SD cards sold is counterfeit. Those cards are not actually produced by the brand that is printed on it.
Which makes RMA'ing cards also a bit problematic. Shop you bought it from may be closed due to lockdown, and cannot send it to manufacturer either if it turns out they did not actually produce the card...

closed 2021-01-07 by maxnet

lurch, 2021-01-07:
@davidplante There's some tips [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives) which may help?

davidplante, 2021-01-08:
Didn't help much. Before Micro SD card showing 2TB for the size instead of 64GB, it showed something like 64MB. After reading a couple of post over the Net, I found a lot of people having this behaviour after writing Raspberry over a Micro SD card. My problem is that I may have messed too much my card and now it's seems broken.
I should have read more before messing with Disk Management to adjust the size to 64GB.

davidplante, 2021-01-08:
@lurch @maxnet I found the problem. Pretty nasty one, but the Micro SD is working very fine now.
After visiting a lot of forum, I saw a lot of people having difficulties with their Micro SD size detection by Windows following a format after installing a raspberry pi Image.
https://www.youtube.com/watch?v=MvheOgqz9x0
This guy had the problem back in 2017
I tried one of the solution mentionned; formatting the SD card using a phone. Worked first try!
Now, Windows sees the card at 64GB. Etcher and RP_Imager still sees the card at 2TB, using the same cheap reader, witch wrote at 0.8MB/S.
Now, I use the usb reader provided with San Disk for my Micro SD. All my problems are solved. 64GB detected, writing at 19.1 MB/S.
Writing with Etcher at the moment. Will try after with Raspberry Pi Imager.

maxnet, 2021-01-08:
Instead of formatting with a phone, you can also use the “format as FAT32” option in Imager if you no longer want to use the card for RPI OS, but want to use the full capacity in Windows again.

(That will not work against cards already appearing larger than they should be. So that is why I didn’t suggest it for your 2 terrabyte card problem)

shin-san, 2021-01-08:
@davidplante, I guess I'll share a bit of my experience.

I played around disk management and formatting with both my SSD and SD Card and used Etcher/imgFlasher for flashing.

I've flashed/un-flashed images using different tools on both drives and have not experienced any bricking.

If you ever need to unflash, don't be scared to just delete the partition if need be. The worst case is the disk becomes unallocated, in which case all you need to do is:

Assuming you are using Windows
1. Initialise the disk (GPT seems to work OK) so it becomes recognisable
2. Right click in the unallocated disk and choose New Simple Volume
3. Format it as exFAT and allocate all available space instead of NFTS. This in turn will allow you to easily flash images using all image flashers available out there.
4. Flash it with the image flasher tool of your choice

Note: you can keep doing this as many times you like until you become familiar with how things work. There are instances where the drive may look like it is corrupted after flashing, but it is all normal (worked fine plugging into RPi4) as I expect Windows doesn't understand Linux partitions.

Good luck! 👍

lurch, 2021-01-08:
> Etcher and RP_Imager still sees the card at 2TB, using the same cheap reader, witch wrote at 0.8MB/S.

Sounds like a bug in the firmware of your cheap SD card reader (which is what converts the read/write calls that Windows sends over USB into the actual read/write calls that happen on the SD card). There's obviously nothing that RPi Imager can do about that!

> Now, I use the usb reader provided with San Disk for my Micro SD. All my problems are solved. 64GB detected, writing at 19.1 MB/S.

Good to hear :tada: