raspberrypi/rpi-imager #143: Crash when running under Wayland

B4dM4n, 2021-01-11:
`rpi-imager` currently can't be run under Wayland.

I'm using [sway](https://github.com/swaywm/sway) v1.5.1 with Qt v5.15.2 and `rpi-imager` v1.5.

Running `rpi-imager` in a terminal results in the following output:
```
Not restoring saved window position as it falls outside of primary screen
IP: QHostAddress("192.168.0.109")
xdg_surface@26: error -1: Tried to set invalid xdg-surface geometry
The Wayland connection experienced a fatal error: Protocol error
```

My window manager log then contains the following output:
```
[ERROR] [types/xdg_shell/wlr_xdg_surface.c:282] Client tried to set invalid geometry
```

This indicates that `rpi-imager` tries to create a window with invalid dimensions:

https://github.com/swaywm/wlroots/blob/0.12.0/types/xdg_shell/wlr_xdg_surface.c#L281-L282
```C
	if (width <= 0 || height <= 0) {
		wlr_log(WLR_ERROR, "Client tried to set invalid geometry");
```

Running `rpi-imager` v1.5 under Ubuntu 20.04 in a Wayland session also creates no window and freezes the program indefinitely.

maxnet, 2021-01-11:
Try if it works better if you remove wayland from the "if" statement at: https://github.com/raspberrypi/rpi-imager/blob/d129d12b004f9097d8fd930bb27f7f6b9b3d1d8b/imagewriter.cpp#L49

B4dM4n, 2021-01-11:
Removing the wayland check fixes the problem and shows the window correctly.
```patch
diff --git a/imagewriter.cpp b/imagewriter.cpp
index a6b0a97..0c54c25 100644
--- a/imagewriter.cpp
+++ b/imagewriter.cpp
@@ -46,7 +46,7 @@ ImageWriter::ImageWriter(QObject *parent)
     connect(&_polltimer, SIGNAL(timeout()), SLOT(pollProgress()));
 
     QString platform = QGuiApplication::platformName();
-    if (platform == "eglfs" || platform == "wayland" || platform == "linuxfb")
+    if (platform == "eglfs" || platform == "linuxfb")
     {
         _embeddedMode = true;
         connect(&_networkchecktimer, SIGNAL(timeout()), SLOT(pollNetwork()));
```

commit 2844b5bd (2021-01-11): Do not activate embedded mode for wayland

Embedded (fullscreen) mode was activated automatically if QPA
platform type is eglfs, wayland or linuxfb.
However although setting window size to -1 does result in fullscreen
on eglfs, on wayland it results in
"Client tried to set invalid geometry"

Do not activate embedded mode for wayland.

Closes #143
closed 2021-01-11 by maxnet