raspberrypi/rpi-imager #146: WARNING: cgroup v2 is not fully supported yet

oldviking, 2021-01-13:
Add cgroup v2 support

maxnet, 2021-01-13:
You did compile the application from source with "cmake ." -> "make" -> "make install"?

As opposed to using some third-party snap/appimage/flatimage/other-container-technology-of-the-week?

oldviking, 2021-01-13:
Tomorrow I will compile the program, I saw this is a common Problem of snap

maxnet, 2021-01-17:
>Tomorrow I will compile the program, I saw this is a common Problem of snap

Snap problems can be reported here: https://github.com/popey/imager-snap/issues

Closing this.
Reopen if you can reproduce the problem with a normal compile. (Don't think so, because the error talks about confinement and we are not doing any).

closed 2021-01-17 by maxnet