raspberrypi/rpi-imager #147: imager_1.5.exe is flagged having an Malware by VirusTotal

yoonghm, 2021-01-16:
imager_1.5.exe is flagged by VirusTotal having a malware (with 80% confidence):

https://www.virustotal.com/gui/file/e150b14b8129257ade06cbed91124b213896c98e5c3c85fe540d0f2028a7dbac/detection

maxnet, 2021-01-16:
>However, I think it is a false positive, meaning that it does not have a malware.

Yeah, we haven't had a single release without some scanner giving a false positive.
In each case it only detected something in our installer, not the program itself.

Think the problem with Virustotal is that they add every virus scanner in the world to their site.
Some of those unknown companies do a very bad job at making virus signatures.
We use the NSIS installer (from the original makers of WinAMP) to install the software. The same installer software is used by A LOT of other Windows software programs.

And yes, probably the installer is also used by some malware.
So if an anti-virus company spots some malware that installs itself using the same NSIS installer software, and it does a bad job making a virus signature, not taking something that is unique to the malware, but taking the signature of a bit of code that is always in the NSIS installer, then that can and will flag us as well.

Problem is also that those unknown virus scan companies are kinda hard to reach.
The last one I tried to contact had a bouncing e-mail address that you were supposed to use to submit white list requests, and an online form to submit samples that would only accept files up to 10 MB while our installer is larger than that...

maxnet, 2021-01-16:
Reopening this.

Where did you get that file from?

Your imager-1.5.exe has 256sum e150b14b8129257ade06cbed91124b213896c98e5c3c85fe540d0f2028a7dbac and the virustotal page claimed it is not signed

Signature Info
Signature Verification
 File is not signed

While our imager-1.5.exe has sha256sum 9c62170a4180d64dfffc6f2af4bc5f3dc04cafd99c2511ed3af38dad9e2173e2

maxnet, 2021-01-16:
Ok, seems you had a truncated download.
I managed to "reproduce" your sha256 hash by truncating the original file down to 18470576 bytes to match the virustotal report.

```
$ sha256sum imager_1.5.exe
9c62170a4180d64dfffc6f2af4bc5f3dc04cafd99c2511ed3af38dad9e2173e2  imager_1.5.exe
$ dd if=imager_1.5.exe of=imager_truncated.exe bs=1 count=18470576
18470576+0 records in
18470576+0 records out
18470576 bytes (18 MB, 18 MiB) copied, 15.9637 s, 1.2 MB/s
$ sha256sum imager_truncated.exe 
e150b14b8129257ade06cbed91124b213896c98e5c3c85fe540d0f2028a7dbac  imager_truncated.exe
```

So no malware.
Just download it again.

closed 2021-01-16 by maxnet

yoonghm, 2021-01-20:
You are correct that the file was truncated as it could run but terminated by Windows OS.