raspberrypi/rpi-imager #148: No error feadback received

Kohoshi12, 2021-01-17:
I tried to install a Raspberry Pi OS 64 bits image with error, but instead of an error message, the level of the burning went over 200%. On Raspberry Pi OS, Imager 1.5.

lurch, 2021-01-18:
Could you explain in more detail what you mean by "I tried to install a Raspberry Pi OS 64 bits image with error" ?

What SD card are you using? What SD card-reader are you using? Can you confirm the exact filename of the image you were trying to flash?

Kohoshi12, 2021-01-18:
I used a 64 Gb sd card. The image was smaller than it should be when it's extracted. At the moment I won't be able to provide any more information about the sd card and the reader.

maxnet, 2021-01-18:
>The image was smaller than it should be when it's extracted

What kind of image?
Is it a .zip with more than one file inside?

Kohoshi12, 2021-01-18:
It was a zip file with a image file inside.

maxnet, 2021-01-18:
Does the issue only occur with that specific .zip file?
Writing images does work correct (and does not go beyond 100% progress) with that SD card reader and SD card, if you choose any of the standard images (such as normal RPI OS) from the list offered by Imager?

Kohoshi12, 2021-01-18:
I'll have to test that later. I was thinking about making the usb port being able to boot a system to solve that, but that's probably nosense. I have no problem to use the same program, sd card and reader on my x86 pc though.

Kohoshi12, 2021-01-18:
I actually had tested the zip file of the same OS and also the image file, the burning process gave me a proper error message.

Kohoshi12, 2021-02-07:
The sd card used is 64GB, XC I (10). The sd card I was using does not work as it should, I tried to burn Twister OS on it, a Raspbian based OS, and it did not work. But using the Raspberry Pi 4 for burning either Twister OS and Manjaro Arm worked well. The problem was really my sd card.

cillian64, 2023-10-19:
Sounds like this is resolved

closed 2023-10-19 by cillian64