raspberrypi/rpi-imager #149: Kali linux ARM images write

KerteszDomokos, 2021-01-21:
Hi,
I use the Imager, and I have a problem.
I would use Kali linux in RPi. I write with  Raspberry PI Imager the .img.xz files into SD card. The error is this:

[image]

If reconnecting the card, the computer isn't sens the disk.
If I unpack the .xz file to .img, I get unchanged error.
The image is the official Offensive Security file, from [https://www.offensive-security.com/kali-linux-arm-images/](here), and I use the 2020.4 version. 
If you must for help more informations, I'm available in everything.

System information:
Windows 10.0.18363.959
Rpi Imager v1.2
SD card: standard, 32gb.

Sorry, my English isn't perfect.

Thanks!

maxnet, 2021-01-21:
You downloaded the image to a folder that has special characters in it.
Does the problem also exist if you put it in a folder with only regular ascii characters in it? (g:\abc)

maxnet, 2021-01-21:
>Rpi Imager v1.2

Also upgrade to v1.5 first

KerteszDomokos, 2021-01-22:
Thanks, in the v1.5 version works well.
The path isn't problem in this version.

I dont' test the path problem with v1.2.

maxnet, 2021-01-22:
Good to hear it does work in the newest version.
Was already fixed by fa7637e7dc89eea896e2b6b229f570f52c5324e3

closed 2021-01-22 by maxnet