raspberrypi/rpi-imager #152: I know this keeps coming up

Sageious, 2021-02-04:
So I have gone through the issues section and I have notice that this keeps happening. Micro SD cards getting unreadable or they can't be used. I am new to all this but ever since I tried to download Lubuntu for the first time or maybe it was when I did a Kali in a bootable USB every image I have done since then will not stick to the SD card or the SD card is no longer usable. I have gone through 3 myself. The Kali and Lubuntu were custom Iso however I pulled them from the source and no where else. I have also noticed that when I put my micro sd cards in it pulls up two letter drives instead of the just one which I feel might have been the reason of breaking one my of Micro SD cards.

maxnet, 2021-02-04:
Recall Lubuntu only made an image for the Pi 2 at some point in the past, and did not bother to keep it updated afterwards.
If you got a more recent Pi model chances are that indeed is not going to work...

If you mean by "no longer usable" that Windows no longer sees the full capacity of the SD card, try the "format card as FAT32" option.

closed 2021-02-04 by maxnet

Sageious, 2021-02-04:
So when I try to reformat the card it only tries to format 1,1 gb the rest is cannot be detected.

maxnet, 2021-02-04:
Start Imager, press "choose OS" and select "format card as FAT32"
Do NOT use the format option Windows itself provides.