raspberrypi/rpi-imager #155: No matter what changes I make, the installation remains the same

revantjhalani, 2021-02-11:
I've tried making changes to the source files by cloning exactly as asked for in the readme. 
After doing so, I build, resulting in a 377 KB file for Ubuntu which I can then install. 
But no matter what I do, the end result is always default. 
I remove the installation using sudo apt --purge remove rpi-imager and then reinstall the new package/ 
What am I doing wrong.? 
In order to test, I replaced all instances of "CHOOSE OS" and built again but it still remians the same so I have no clue what I'm doing wrong at this point./ 
Any help would be appreciated, thanks.

maxnet, 2021-02-11:
During development I suggest you test without installing first.
All files it uses get embedded in the rpi-imager executable, so can just build it locally -without using the .deb packaging stuff-, and execute it from the local directory to see if things work.

```
mkdir mybuilddirectory
cd mybuilddirectory
cmake ..
make
./rpi-imager
```

If that does work properly, you can take a look at what is wrong with the .deb stuff.
Maybe it has leftovers from a previous compilation attempt, and you need to "debuild -T clean" or anything.

revantjhalani, 2021-02-11:
okay so, I created another VM to see if it was something to do with the VM or the uninstallation, and it seems to be the latter so sorry to bother you with this, but do you know if there are any remnant files that are causing this, 
sudo apt-get purge --auto-remove rpi-imager
sudo dpkg --remove --force-remove-reinstreq rpi-imager
sudo dpkg --purge --force-depends rpi-imager
I have tried all three of these, and none work to make it so that the new installation is completely fresh.
Also, thank you for telling me how to run it from source, it's a godsend :) 
Thank you

maxnet, 2021-02-11:
It should not be necessary to remove anything with apt-get.
Installing the new package will overwrite /usr/bin/rpi-imager and that is the only file that matters (all qml files, images, etc. are embedded inside there)

If it is using old files it was likely something inside the build directory not getting rebuild, when it should.
Can happen due to various reasons, e.g. if your system has incorrect date/time/timezone it may think the compiled version is newer than source.

closed 2021-02-11 by maxnet