raspberrypi/rpi-imager #156: U3 should be removed

mobluse, 2021-02-22:
I erased a SanDisk USB stick that had U3 smart, but that didn't erase the U3 partition which is necessary to remove in order to use the entire drive. I had to remove it using:
sudo u3-tool -p 0 /dev/sda
That worked in latest Raspberry Pi OS Buster on Raspberry Pi 4B. My suggestion is that this u3-tool is included and used to remove U3 partitions.

lurch, 2021-02-22:
It might not be "safe" for RPi Imager to blindly assume that the user wants their U3 partition to be wiped? :man_shrugging:

mobluse, 2021-02-24:
I believe that if you choose Erase in Raspberry Pi Imager you expect the entire card/stick to become FAT32 (as it is now), but maybe an extra warning about removing U3 could be issued.

lurch, 2021-02-24:
I don't believe that special-handling for U3 disks is something offered by any other disk-image writing software?

maxnet, 2021-02-25:
https://kb.sandisk.com/app/answers/detail/a_id/5358/~/u3-launchpad-end-of-life-notice/
"U3 technology has reached end of life. SanDisk began phasing out support for U3 Technology in late 2009."

closed 2021-02-25 by maxnet