raspberrypi/rpi-imager #157: Crash on Startup with Linux Mint 20.1

pnlrogue1, 2021-02-24:
Hi there,

While trying to launch imager on my desktop PC today, it just instantly crashed. Tried reinstall over the top/uninstall then reinstall/purge then reinstall, all from .deb downloaded freshly today. It had previously worked on the same desktop PC.

When launching from the desktop, the window starts to draw but then closes. When launching from the terminal, I get this:

```shell
❯ sudo rpi-imager
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior QSurfaceFormat::DoubleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
[1]    151729 abort      sudo rpi-imager
```

My system information:

```shell
❯ neofetch
             ...-:::::-...                 phil@BlackDragon-Mint 
          .-MMMMMMMMMMMMMMM-.              --------------------- 
      .-MMMM`..-:::::::-..`MMMM-.          OS: Linux Mint 20.1 x86_64 
    .:MMMM.:MMMMMMMMMMMMMMM:.MMMM:.        Host: X470 AORUS ULTRA GAMING 
   -MMM-M---MMMMMMMMMMMMMMMMMMM.MMM-       Kernel: 5.4.0-42-generic 
 `:MMM:MM`  :MMMM:....::-...-MMMM:MMM:`    Uptime: 5 hours, 53 mins 
 :MMM:MMM`  :MM:`  ``    ``  `:MMM:MMM:    Packages: 3689 (dpkg), 50 (brew), 27 (flatpak), 17 (snap) 
.MMM.MMMM`  :MM.  -MM.  .MM-  `MMMM.MMM.   Shell: zsh 5.8 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM-MMM:   Resolution: 2560x1440, 2560x1440 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM:MMM:   DE: Cinnamon 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM-MMM:   WM: Mutter (Muffin) 
.MMM.MMMM`  :MM:--:MM:--:MM:  `MMMM.MMM.   WM Theme: Mint-Y-Dark (Mint-Y-Dark) 
 :MMM:MMM-  `-MMMMMMMMMMMM-`  -MMM-MMM:    Theme: Mint-Y-Dark [GTK2/3] 
  :MMM:MMM:`                `:MMM:MMM:     Icons: Mint-X-Dark [GTK2/3] 
   .MMM.MMMM:--------------:MMMM.MMM.      Terminal: HyperTerm 
     '-MMMM.-MMMMMMMMMMMMMMM-.MMMM-'       Terminal Font: MesloLGS NF 
       '.-MMMM``--:::::--``MMMM-.'         CPU: AMD Ryzen 7 2700X (16) @ 3.700GHz 
            '-MMMMMMMMMMMMM-'              GPU: NVIDIA GeForce GTX 1070 
               ``-:::::-``                 Memory: 8942MiB / 16020MiB
```

maxnet, 2021-02-24:
>❯ sudo rpi-imager

Same error if you run without sudo?
(doesn't need that, it outsources privileged actions like opening the disk device to udisks2).

Other OpenGL applications like glxgears (from mesa-utils) do work properly?

pnlrogue1, 2021-02-24:
Yep - identical

maxnet, 2021-02-24:
>I'm not familiar with that application or what the output should look like

It is a simple test application, which should display a couple turning cogs:

[image]

If it does not, then your graphics is screwed up.
That is not an Imager specific problem then.
More something to ask about on Mint/Nvidia forum.

pnlrogue1, 2021-02-24:
Following a reboot, all was well (and I see the gears when I run glxgears). Turns out that it wasn't anything to do with Imager! Hope this thread will help someone else in the future!

Message for posterity: Even Linux sometimes needs turning off and on again!

closed 2021-02-24 by pnlrogue1