raspberrypi/rpi-imager #158: Can't write to an NVME device

peterharperuk, 2021-02-25:
I've got a cm4 / io board and you can plug a NVME drive in this and use it for rootfs.
It would be nice if you could use rpi-imager to write an OS to it but the drive doesn't appear in the list.

Looking more closely this seems to be because it's marked as non-removable which causes "isSystem" to be true.
Would it be reasonable to relax this rule? The drive has no mount points, isn't used for the "system" so should be a candidate for an install?

peterharperuk, 2021-02-25:
$ lsblk --bytes --json --paths --output-all | grep nvme
      {"name":"/dev/nvme0n1", "kname":"/dev/nvme0n1", "path":"/dev/nvme0n1", "maj:min":"259:0", "fsavail":null, "fssize":null, "fstype":null, "fsused":null, "fsuse%":null, "mountpoint":null, "label":null, "uuid":null, "ptuuid":"7489641a", "pttype":"dos", "parttype":null, "partlabel":null, "partuuid":null, "partflags":null, "ra":128, "ro":false, "rm":false, "hotplug":false, "model":"SK hynix BC501 HFM128GDJTNG-8310A", "serial":"NI91T004112301715", "size":128035676160, "state":"live", "owner":null, "group":null, "mode":"brw-rw----", "alignment":0, "min-io":512, "opt-io":0, "phy-sec":512, "log-sec":512, "rota":false, "sched":"none", "rq-size":1023, "type":"disk", "disc-aln":0, "disc-gran":512, "disc-max":2199023255040, "disc-zero":false, "wsame":0, "wwn":"eui.ace42e0090021a1a", "rand":false, "pkname":null, "hctl":null, "tran":"nvme", "subsystems":"block:nvme:pci:platform", "rev":null, "vendor":null, "zoned":"none"}

maxnet, 2021-02-25:
Not sure if whether something is mounted is a good criteria to determine suitable destination drives.
On many multi-boot systems the "other operating system" is not mounted either.
Many Linux distributions do auto-mount every removable drive with a file system though.

If you are an advanced user and would like more destination drive choices change "#define DRIVELIST_FILTER_SYSTEM_DRIVES" to "false" in config.h and recompile.

peterharperuk, 2021-02-26:
Yes DRIVELIST_FILTER_SYSTEM_DRIVES works although it gives me the option to overwrite my current rofs partition which might be interesting. But it would be nice to have an option like this without a recompile. I was most interested in solving this for when running on Raspberry PI OS. Currently it doesn't seem possible to use the imager app to install an OS on an NVME drive,

lurch, 2021-02-26:
AFAIK the problem isn't that it's an NVMe drive, merely that it's considered as an "internal" drive. And RPi Imager filters them out in order to prevent the user accidentally trashing the harddrive / SSD they've booted from.
I think if you put your NVMe drive into a USB enclosure then RPi Imager would then see it as an "external" drive and have no problem writing to it?

peterharperuk, 2021-02-26:
Yes - if I buy an M2 USB enclosure then it works. It's just a bit annoying that I'd have to buy this piece of kit just to do that. Another solution is to use piclone. It just seems a bit annoying that I can't install Raspberry PI OS on an M2 disk (that isn't even mounted) on Raspberry PI OS.

lurch, 2021-02-26:
> Another solution is to use piclone.

Or good old `dd` :wink: 

> It just seems a bit annoying that I can't install Raspberry PI OS on an M2 disk (that isn't even mounted) on Raspberry PI OS.

The only way to "solve" that annoyance would be to make RPi Imager more "dangerous" for the 99.99% of people who _aren't_ running it on a CM4 with an M2 SSD; and I suspect that's a trade-off that the RPi Imager developers are unwilling to make.

peterharperuk, 2021-03-09:
Sorry, didn't notice you'd fixed this. Just tested it and I can now choose an nvme drive as the destination. Thanks!

closed 2021-03-09 by peterharperuk