raspberrypi/rpi-imager #159: Unable to use custom image on Fedora 33

ethanholz, 2021-03-05:
I attempted to write a custom .img file and when I selected it with Nautilus, I am returned to the menu to select an image but the custom image is not selected nor is there any error in the command line.

maxnet, 2021-03-05:
You did compile rpi-imager from source?
(As opposed to using any sandboxed environment like snaps)

ethanholz, 2021-03-06:
@maxnet that is correct, compiled from the repo. If I call the imager  from the command line and pass it the .img file, that allows for me to flash an SD card with the image; I just can't from the GUI directly n

maxnet, 2021-03-06:
Can you try compiling from latest git again, and see if it works better now?

Think the code we had to workaround monitors with broken EDID caused problems on Fedora.
Reverted it.

ethanholz, 2021-03-06:
Yup, that worked! Thank you for taking a look at this!

maxnet, 2021-03-06:
Good to hear it works now.

closed 2021-03-06 by maxnet