raspberrypi/rpi-imager #160: Raspberry Pi Imager UI opens with wrong language

keelaro, 2021-03-10:
I have installed the RPi Imager on MacOS Big Sur 11.2.3 and the language that displays is not English even thought this is my default language. 

Having google the issue I saw a suggestion to check the setting i preferences. I noticed that although English is the preferred and default language German and Dutch and Hungarian option were also in my list (there for other purposes). When I delete these from my preference list RPi Imager now opens in English language.

maxnet, 2021-03-10:
Which Imager version are you using?
As this was supposed to be fixed in Imager v1.4

https://github.com/raspberrypi/rpi-imager/issues/81

maxnet, 2021-03-12:
Am afraid I am unable to reproduce the problem.
Only becomes Dutch for me when I drag it to the top in the list of languages.

[image]

[image]

Using v1.5 from raspberrypi.org

maxnet, 2021-03-26:
@keelaro can you try if this one works better for you?

[Raspberry Pi Imager 1.7.zip](https://github.com/raspberrypi/rpi-imager/files/6212737/Raspberry.Pi.Imager.1.7.zip)

maxnet, 2021-03-29:
Fixed in 1.6.1

closed 2021-03-29 by maxnet