raspberrypi/rpi-imager #161: Feature : find raspberry pi's IP

r4ph0u, 2021-03-10:
It could be great to add a feature for finding the raspberry-pi on the local network.

The following guides gives 2 methods for that, if they could be implemented as a simple button that would return all raspberry pi's IP found, it would be great for quick startup (having to access the router's interface makes it less beginner-friendly)

[How to Find Your Raspberry Pi's IP Address](https://howchoo.com/pi/find-your-raspberry-pis-ip-address)

maxnet, 2021-03-10:
Don't think that belongs in Imager.

Why not just set a different .local host name for each Pi from the advanced settings?

Slyke, 2021-03-22:
You could install a DDNS service on the RPi?

maxnet, 2021-03-22:
>You could install a DDNS service on the RPi?

Avahi is installed by default on RPI OS.
And you can the hostname that uses with the advanced options available under SHIFT+CTRL+X in Imager.

closed 2021-03-22 by maxnet