raspberrypi/rpi-imager #162: [OS Request] OctoPi

guysoft, 2021-03-11:
Hey,
I am the maintainer of OctoPi, and a few other distros since I am also the creator of CustomPiOS. So this is the first attempt our of expected several.
I got a request to add OctoPi to rpi-imager on my issue tracker https://github.com/guysoft/OctoPi/issues/726

I can't see any docs on how to add an image. So I am copying more or less what I saw in #83
```json
{
    "os_list": [
        {
            "name": "OctoPi",
            "description": "A Raspberry Pi distribution for 3d printers. It includes the OctoPrint host software for 3d printers out of the box and mjpg-streamer with RaspiCam support for live viewing of prints and timelapse video creation.",
            "icon": "https://raw.githubusercontent.com/guysoft/OctoPi/devel/media/OctoPi.png",
            "subitems_url": "http://unofficialpi.org/rpi-imager/rpi-imager-octopi.json"
        }
    ]
}
```

And the json file is in place:
http://unofficialpi.org/rpi-imager/rpi-imager-octopi.json

Note: We are expecting next release to also include a 64bit builds which is already in the nightly. If there is a section to add in advance it might also be helpful

ghollingworth, 2021-03-11:
The icon needs to be 40x40 and the text needs to be limited to fit into the box.  You can test it with rpi-imager by copying a local version of the global os_list_imagingutility.json and running it with the --repo command-line option:

C:\Program Files (x86)\Raspberry Pi Imager>rpi-imager.exe --repo os_list_imagingutility.json

[image]

guysoft, 2021-03-11:
Fixed json in the original request.

Here is how it looks.

Also I warmly recommend adding this to the README file in the repo, or to the wiki page on this githubl, otherwise I expect you will be repeating yourself.

[image]

[image]

ghollingworth, 2021-03-11:
Does the one image support all Raspberry Pi devices?  In which case I'll probably add it without the submenu

guysoft, 2021-03-11:
All Raspberry Pi devices are supported.

ghollingworth, 2021-03-11:
Ok, approved for adding, should appear sometime today...

guysoft, 2021-03-11:
Looks like its there now, thanks!

I hope to automate the json stuff here now and also post other CustomPiOS popular distos.

[image]

guysoft, 2021-03-15:
@ghollingworth Hey, sorry to bug again, but @foosel has pointed out to me that the url should be HTTPS and not HTTP in order to avoid man-in-the-middle attacks.

So it should be:

```
{
    "os_list": [
        {
            "name": "OctoPi",
            "description": "A Raspberry Pi distribution for 3d printers. Ships OctoPrint out-of-the-box.",
            "icon": "https://raw.githubusercontent.com/guysoft/OctoPi/devel/media/rpi-imager-OctoPi.png",
            "subitems_url": "https://unofficialpi.org/rpi-imager/rpi-imager-octopi.json"
        }
    ]
}```

XECDesign, 2021-03-16:
Should be using HTTPS now.

guysoft, 2021-03-16:
Works, closing

closed 2021-03-16 by guysoft