raspberrypi/rpi-imager #164: RPi imager taking too much time to write on SD card

ajita02, 2021-03-17:
I am using samsung SD adapter for SD card. I have 32GB sandisk ultra class 10 micro SD card. I tried to flash raspbian using rpi imager but its taking too much time like 37% write completion after 1 hr. Is there any other way to flash raspbian in SD card?

maxnet, 2021-03-17:
"Samsung SD adapter" is an external USB SD card reader?
Or do you mean a micro SD <-> full size adapter like this:

[image]

that you use to put the card into the internal SD card reader of a laptop?

If it takes a long time to write it is also hard to tell if there is something wrong with the card, the USB card reader, the card adapter, or if the mirror the Imager program is downloading the operating system image from has issues.
You can rule out the latter causing problems by downloading RPI OS to disk first from: https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-01-12/2021-01-11-raspios-buster-armhf.zip
And selecting "choose file" in Imager.
(No need to extract the file, it can handle .zip itself).

ajita02, 2021-03-18:
Hi @maxnet , I am using same one shown in image which I used as micro SD to SD adapter. Thanks for providing link. I will try to flash by downloading zip file.

ajita02, 2021-03-23:
Hi @maxnet , I uninstalled rpi imager and then installed again. It worked for me without using zip file.

closed 2021-03-23 by ajita02