raspberrypi/rpi-imager #167: Create a separate repository for language files only

bovirus, 2021-03-18:
@maxnet 

Could be a good idea move only the language files on a separate repository.

Then the translators can set to have a notification for changes in repository where are avaiilable only language files.

tdewey-rpi, 2024-02-06:
Circling around to this - I'm not sure this would produce the intended effect - I could instead see a world where people forget to make a PR for the language files and we wind up with untranslated strings all around.

I don't believe there's a particularly good flow for this in the GitHub-verse. A CODEOWNERS file or similar would be good, and I'll have an investigation on if there's a github extension that supports this, but without that manually tagging translators might be the best path forward.

Closing out - it's not a pending issue with rpi-imager, but I'm happy to accept counterpoints.

closed 2024-02-06 by tdewey-rpi

bovirus, 2024-02-06:
@tdewey-rpi 
Other Github project create separate repository only for language files.
In this way for most active translators you can assign right to upload directly language file without need a PR.

tdewey-rpi, 2024-02-06:
Thanks for the response, @bovirus

Unfortunately, that is also undesirable. Raspberry Pi Imager carries the Raspberry Pi branding, and as such we're exposed to brand damage in the event of translation abuse - an attack that has been executed recently in high-profile environments.

With the repository structured this way, maintainers get the ability to perform loose verification of the translation ahead of merge and release - allowing us to avoid such attacks.