raspberrypi/rpi-imager #172: Fedora package

AquaL1te, 2021-03-19:
As per the RPM package guidelines, I'm informing you that I've prepared a Fedora package of your project and it's under review at the moment: https://bugzilla.redhat.com/show_bug.cgi?id=1940937

maxnet, 2021-03-23:
Note that custom file selection is known broken on Fedora in 1.6
See: https://github.com/raspberrypi/rpi-imager/issues/159
Apply https://github.com/raspberrypi/rpi-imager/commit/2d87a2275f9a2525f000d0f216468b9cced97e98.patch

AquaL1te, 2021-04-08:
Package is now in Fedora stable (with the patch included mentioned by @maxnet).

closed 2021-04-08 by AquaL1te

lurch, 2021-04-08:
In the meantime, a newer version of RPi Imager has been released https://github.com/raspberrypi/rpi-imager/releases/tag/1.6.1

AquaL1te, 2021-04-08:
@lurch, thanks for the heads-up! I guess the [release monitoring](https://release-monitoring.org/project/185260/) of Fedora only kicks in once the packages the stable branch or something. I didn't got a notification. I'll update the package and I trust it will keep me informed going forward.

lurch, 2021-04-08:
You could also subscribe to release-notifications on this repo if you wanted to :wink: 
https://github.blog/changelog/2018-11-27-watch-releases/

AquaL1te, 2021-04-08:
Good advice! :)

New updates are open for QA review and will be in stable in 7 days if all goes well.
* https://bodhi.fedoraproject.org/updates/FEDORA-2021-5926f0a056
* https://bodhi.fedoraproject.org/updates/FEDORA-2021-fc63def3f8