raspberrypi/rpi-imager #173: Feature Request - apt upgrade

CharlesGodwin, 2021-03-20:
Love v1.6. But....
Could you add an advanced feature to enable an automatic upgrade during setup? Essentially add these two lines to the firstrun.sh file

`sudo apt update`
`sudo apt -y full-upgrade`

Then the first user experience will be an up to date OS.

Perhaps this should not be an advanced option but for all users. Being up to date is one less problem for new comers.

lurch, 2021-03-21:
I don't think it makes sense to run those commands automatically, because the user may not have configured an internet connection (e.g. typing in their wireless network password details) when their Raspberry Pi boots up for the first time? (and there's obviously no way for RPi-Imager to know if the user is going to be connecting over wireless or over ethernet!)

The "First Run Wizard" (written by @spl237 ) that you get when you first view the Pi's graphical desktop **does** do these steps though (after helping the user set up their internet connection).

maxnet, 2021-03-21:
>e.g. typing in their wireless network password details

Eh, why would the user be forced to type in difficult passwords, when the "advanced options" functionality can just borrow the wifi password from the computer running Imager (well, currently limited to Windows)? :-)

Not sure if offering apt upgrading is a good solution though.
Personally I like the concept of providing images with all updates provided better. Aka http://downloads.raspberrypi.org/nightlies/

lurch, 2021-03-21:
> Since I learned the GUI version does this anyway, it should only remain an option for people like me who normally use the Lite image.

I suspect that the Venn-diagram overlap of people who choose to use the Lite image but don't know how to run an `apt upgrade` is probably fairly small? :wink:

CharlesGodwin, 2021-03-21:
I agree, I am suggesting it only as a convenience. I like this setup approach as it provides a ready to run image. I have a script now that does a lot of this now, and plan to revise it to suit this new capability.

closed 2021-03-26 by CharlesGodwin