raspberrypi/rpi-imager #174: Feature Request - Run script on first (or every) boot

Slyke, 2021-03-21:
The new changes on the latest version are awesome, but it would be extremely useful to be able to include a single sh file that will be run on the first boot (or every boot if that's easier) of the pi.

Right now I have a script that DDs the disk, write RPiOS to it, unmount, remount, then add some stuff into rc.local:

```
  LCLRC=$ROOTFSDIR/etc/rc.local
  echo "Writing init script"
  # Wait for mount
  while [ ! -f $LCLRC ]
  do
    printf "."
    sleep 1
  done

  sed -i -e "$(grep -n 'exit 0' $LCLRC | cut -f1 -d: | tail -n 1)d" $LCLRC
  echo "" >> $LCLRC
  echo "touch /tmp/rc.local_out.txt" >> $LCLRC
  echo "whoami > /tmp/rc.local_out.txt" >> $LCLRC
  echo "if [ -e /boot/setup.txt ]; then" >> $LCLRC
  echo "  wget -O - MY_SCRIPT_URL_FOR_ROOT  | bash >> /tmp/rc.local_out.txt" >> $LCLRC
  echo "sleep 5" >> $LCLRC
  echo "  wget -O - MY_SCRIPT_URL_FOR_RPI_USER | su - $PIUSR -c bash >> /tmp/rc.local_out.txt" >> $LCLRC
  echo "  rm /boot/setup.txt" >> $LCLRC
  echo "else" >> $LCLRC
  echo "  echo \"init.sh not run due to no /boot/setup.txt file\" >> /tmp/rc.local_out.txt" >> $LCLRC
  echo "fi" >> $LCLRC
  echo "" >> $LCLRC
  echo "exit 0" >> $LCLRC
```

Which works most of the time, but sometimes Ubuntu doesn't remount the drive correctly after DD has finished, or a bazillion other things that go wrong.

maxnet, 2021-03-21:
>Which works most of the time, but sometimes Ubuntu doesn't remount the drive correctly

Well, we only support "advanced options" for RPI OS at the moment.
Logic assumes the FAT partition will be mounted after boot at /boot. And think Ubuntu has it at an alternative location (/boot/efi ?)

lurch, 2021-03-22:
See also the big discussion in https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/40

danielo515, 2021-05-26:
I see that this tool already adds a firstboot script, so just allow people to append stuff to that script?

Slyke, 2021-05-26:
> I see that this tool already adds a firstboot script, so just allow people to append stuff to that script?

Was this added very recently? I recently downloaded it and I don't see this option.

tdewey-rpi, 2024-02-06:
Closing as stale.

As a previous commenter mentioned, if you _really_ want to do this, you could consider appending to the firstboot.sh script that is put in the boot partition as part of image customisation (on a customised OS).

Note, however, that script is run _before the network comes up_, so may be of limited utility, and support for that mechanism is not conserved across all OS' offered by the Raspberry Pi Imager OS list.

closed 2024-02-06 by tdewey-rpi