raspberrypi/rpi-imager #175: Raspberry pi logo doesn't have to be that big. Make it work and fit fine on lower resolutions

mgrouch, 2021-03-22:
Raspberry pi logo doesn't have to be that big. Make it work and fit fine on lower resolutions. With resolution 800x480 takes too much screen 
space and header goes under desktop toolbar. There are only few buttons and a logo. It should be made usable  even with smaller screen
resolutions.

maxnet, 2021-03-23:
You may have better luck running Imager fullscreen under console, instead of under X.

```
sudo raspi-config -> system options -> boot -> console autologin
*reboot*
sudo QT_QPA_PLATFORM=eglfs QT_QPA_EGLFS_ALWAYS_SET_MODE=1 rpi-imager
```

mgrouch, 2021-03-23:
Will console version work with touchscreen?

mgrouch, 2021-03-23:
> Will console version work with touchscreen?
>
> Only if you do not have to enter any text.
> No virtual keyboard is present by default.

So it’s not an option

lurch, 2021-03-24:
But I guess you'd need to have a USB keyboard plugged in anyway in order to type the `sudo QT_QPA_PLATFORM=eglfs QT_QPA_EGLFS_ALWAYS_SET_MODE=1 rpi-imager` command? :confused:

mgrouch, 2021-03-24:
Correct.

lurch, 2021-03-24:
Perhaps I'm missing something obvious, but why is it so important for (the advanced options of) RPi Imager to work on a touchscreen without plugging in a USB keyboard? Isn't RPi Imager something that you're likely to run fairly infrequently? :shrug: 

P.S. Please take my comments with a grain of salt, I don't work on RPi Imager myself; I'm just curious.

cillian64, 2023-10-19:
The logo is made smaller in 1.8.0, but on crazy-low-resolution displays the GUI is never going to work very well.

closed 2023-10-19 by cillian64