raspberrypi/rpi-imager #177: Window not resizable

Timmmm, 2021-03-22:
It's pretty annoying that neither the window nor the fake "choose OS" dialog can be resized.

[image]

Looking at the QML it seems like there are fixed sizes all over the place unfortunately, so this may be hard to fix. As a temporary workaround you could at least change the size of the dialog so you don't have to scroll.

ghollingworth, 2021-03-22:
This is by design, mostly because then it is exactly the same GUI on all operating systems.

closed 2021-03-22 by ghollingworth

lurch, 2021-03-22:
LOL, this issue is asking for the window to be made bigger, and #175 is asking for the window to be made smaller :wink:

Timmmm, 2021-03-22:
> This is by design, mostly because then it is exactly the same GUI on all operating systems.

Honestly I'm not really sure how to respond to this because the benefits of making windows resizable are so obvious and overwhelming. There are some desktop programs that don't have resizable windows but it's usually because that's just easier to implement. I've never heard of someone *arguing in favour of it!* 🤯 

> this issue is asking for the window to be made bigger, and #175 is asking for the window to be made smaller

If only there was a way to let the user decide how big the window should be...