raspberrypi/rpi-imager #179: Automatically set up SSH & password

Timmmm, 2021-03-22:
Similar to #178, it would be good if there was an option to automatically enable SSH, and set the password during imaging.

Currently you have to Google how to do it and after a few minutes find [a random forum post with instructions](https://www.raspberrypi.org/forums/viewtopic.php?t=169905#p1090844). Then you still have to change the password after logging in. It would be much better if the imager just said

> "What password would you like?" "[...]"
> "Would you like to enable SSH access?" "Yes/no"

ghollingworth, 2021-03-22:
Also...  Ctrl-Alt-X

https://www.raspberrypi.org/blog/raspberry-pi-imager-update-to-v1-6/

closed 2021-03-22 by ghollingworth

lurch, 2021-03-22:
That page says ‘Ctrl-Shift-X’ :wink: