raspberrypi/rpi-imager #180: Make advanced options discoverable

Timmmm, 2021-03-22:
I submitted two issues (#178 and #179) to do automatic setup of SSH and wifi. Turns out that feature already exists, which is fantastic!

Unfortunately it's completely undiscoverable. There's no way to know it exists unless you magically already know.

There should be some visual indication that the option is available - "do you want to set up advanced options?" sort of thing.

lurch, 2021-03-22:
This got discussed in the blog-comments https://www.raspberrypi.org/blog/raspberry-pi-imager-update-to-v1-6/

closed 2021-03-22 by ghollingworth

Timmmm, 2021-03-22:
I have to say you guys make a lot of disappointingly user-hostile decisions. You don't let users resize the window (a very standard feature), you hide the option to set the freaking wifi password. That is so beginner unfriendly!

> This got discussed in the blog-comments

If by "discussed" you mean "almost everyone agreed it was bone-headed", then sure. Look at the comments from your users!

> But why o why continue to make it so hard? A little button ‘Advanced’ was still too much?

> What you call Advanced, most of us call More Options.

> As probably already people suggested, please add some button to open the advanced Settings. I really like being able to set a few things (as Jeff Geerling already stated), but for sure I won’t remember the key combination in a few weeks.

> Mmmmm. A GUI interface with keyboard-only functions. That makes so much sense (not).

> Extremely valuable new optionst they should not be accessible in a so esoteric and only way like an Easter egg (key combination). If a program HAS a feature it has to be exposed. Just appropriately labelled as hard, or for experts or extra, but still exposed.

I'd also like to reply to one of the two (that I counted) replies in favour:

> Advanced options are for advanced users. If you can’t remember CTRL-SHIFT-X then you probably shouldn’t be messing with the advanced settings.

Well for a start [you literally just forgot the shortcut!](https://github.com/raspberrypi/rpi-imager/issues/179#issuecomment-804228941), but secondly how are you supposed to remember the shortcut if you never even knew it existed?

This goes against basically every UX principle ever. I could understand it if the options were super dangerous, but we're talking about setting up wifi here! What world are you living in where you think people with a Raspberry Pi are intimidated by wifi?

Anyway I'll leave you to your crazy designs.

lurch, 2021-03-26:
@maxnet / @ghollingworth Is it worth mentioning the magic advanced-mode key-combo in https://github.com/raspberrypi/rpi-imager/blob/qml/README.md ?

Timmmm, 2021-03-28:
If you actually want users to know about this very useful feature then just put a button for it in the app. Very few people will read the Readme but everyone uses the app.

If you don't want users to know about this feature then you should think hard about why, because it's clearly insane to write a feature that you don't want anyone to actually discover.

lurch, 2021-03-28:
Yes, everybody working at Raspberry Pi is clearly insane :stuck_out_tongue_winking_eye:  Welcome to the madhouse :upside_down_face:

kingosticks, 2021-03-31:
It's not even been 2 weeks and I just had to look up the key combo already. To do that I had to first find the blog post. I'll accept I have a poor memory but that aside, this was by no means resolved in the blog comments. I fully appreciate these extra options might be overwhelming/confusing to our newest users and we absolutely don't want that, but I think there is a middle ground here. Could we discuss what that might look like?

maxnet, 2021-03-31:
>but I think there is a middle ground here

The shortcut sort of was the middle ground.
Between team #we-are-targetting-beginners rejecting the concept of a button, and team #experts coming up with a command line option (which is non-trivial to use on say Mac OS X).

maxnet, 2021-03-31:
>I don't use a Mac so I don't know why that is non-trivial but I'll take your word for it. 

Well, on Mac the application executable is not in the path.
And yes, you do can actually `cd` into .app containers, but I doubt many Mac users know that.
Would require entering something like: `/Applications/Raspberry\ Pi\ Imager.app/Contents/MacOS/rpi-imager --option`

Keyboard combo was also changed "shift" + "ctrl" + "X", to be able to have the same combination on MacOS.
Even though ctrl+alt+x may be easier to remember for some, MacOS lacked "alt".

kingosticks, 2021-03-31:
OK, thanks for that explanation regarding Mac.

So to clarify, is a button with a very clear warning really off the table? Such a button could be small and placed in the bottom corner to reduce the chance of being accidentally pressed. Something like this:

[image]

If someone still feels this could lead to anyone accidentally changing the advanced options, what if one of the advanced options was to always show this button? So then you only have to remember/lookup the magic shortcut once. If that is still somehow disagreeable, you could have the button hidden unless the `--show-advanced-options` command line option is used (for the benefit of non-Mac OS users only).

I'm really struggling to see how this adjustment makes the tool noticeably more confusing for new users. And it's much better for those other users who do want to easily enable SSH, change the hostname, configure WiFi etc.

kingosticks, 2021-04-08:
So, would you accept a pull request with such a change? If not, what could we adjust to make it more apealing to those in charge here?

ghollingworth, 2021-04-09:
For the moment we've already made our decision about how we're going to support this functionality.  As said many times before we do not want to change the user interface at the moment since this is the easiest way for people to image SD cards.

When an easier way appears we may then make this a more complex interface for more advanced users, but until then we will not be changing it.

You are free to build and distribute your own version (as long as you remove the Raspberry Pi logos and wordmark from the front page.)

kingosticks, 2021-04-09:
Disappointing but of course that's your decision to make. Thanks for taking the time to reply.

guysoft, 2022-01-19:
Hey,
> You are free to build and distribute your own version

@ghollingworth

Er - I am actually considering making my own version, since there are about 6 standing ``[OS Request]`` issue with no answers ([filter to requests](https://github.com/raspberrypi/rpi-imager/issues?q=is%3Aopen+is%3Aissue+%22%5BOS+Request%5D%22)).
Does it make sense to you guys there? I don't want to step on anyone toes, though it seems like all other users in the thread might welcome a "unofficialpi" fork.
I am also in the works of adding json generation in CustomPiOS (related https://github.com/guysoft/CustomPiOS/issues/151 ). And knowing what is the stand on that would help me decide if a fork is worth having.

(I was actually got here thinking we could join efforts and make a smoother CI/CD using github actions or something more accessible)

ghollingworth, 2022-01-20:
In general, those requests have sat idle while I think about how best to put them in.  In the end, the more images that do the same thing (home automation or 3D printer for example), I'm just going to add in a secondary list.  For example:

```
Other Specific Purpose OSs
  -> OctoPi
  -> Other 3D printer OSs
      --> ABC
      --> Simply etc
```

guysoft, 2022-01-26:
Hey, so I started looking in to making a fork, for now It already uses its own json list and merges ones in another repo:

https://github.com/guysoft/pi-imager

And it takes it from:
https://github.com/guysoft/pi-imager-web/
Which already has a github action to merge the official .json and add on top of it.

If I get Ci/CD for this running we could have a less restricted and more advanced version of rpi-imager. Perhaps even make it support other SBCs.
It could also be a space to see what works and then merge it up to mainstream.

Timmmm, 2022-01-27:
Nice work guysoft! @Raspberry Pi why did you delete his message? If there's a good reason then you should probably say before people make assumptions. If there isn't then I would strongly consider reinstating it ASAP before people realise!

guysoft, 2022-02-01:
Hey, just to update that I spoke to Gordon and all is good. No need to make any assumptions :)
I might start working on a "packed with distros rpi-imager" once I get OctoPi and FullPageOS RCs out.

kingosticks, 2022-02-01:
Hold the phone, has this feature now been [added](https://github.com/raspberrypi/rpi-imager/commit/8f9fbcffebb2a37c0f9ad0c58e3bddc2789c05a1#diff-d188f41354c75d01783885a9ebda1165b23cdecc402c67e26cd67018de25176cR258-R268) ?

kingosticks, 2022-02-03:
Yeh. It was. Great. Thanks.