raspberrypi/rpi-imager #184: [Feature Request] Inform the user that settings have been successfully saved in Advanced Settings

cl-ement05, 2021-03-23:
On advanced settings, when you save the enabled customizations, nothing happens except for the button which turns a little darker. But this is not a way of saying "settings saved". So it would be great if you could make a toast notification which disappears after some seconds just to inform the user settings have been saved

cl-ement05, 2021-03-23:
> 
> 
> For me (on Linux), when I click the "SAVE" button the options window disappears (which is kinda what I'd expect).

Well this is not happening for me when I change some settings. I am creating an issue

lurch, 2021-03-23:
Ahh, I guess this can be closed in light of #188 ?

cl-ement05, 2021-03-23:
Yeah of course sorry

closed 2021-03-23 by cl-ement05