raspberrypi/rpi-imager #185: Flatpak: cannot flash a custom image

ghisvail, 2021-03-23:
I am forwarding the following [issue](https://github.com/flathub/org.raspberrypi.rpi-imager) from the Flatpak distribution.

The gist of it is that users cannot flash custom images on version 1.6 of Imager. After picking an image file in the file portal, the app stays on the "Choose OS" step instead of moving on to the "Choose Storage" step.

I could not see anything in the logs, so I was wondering whether you could help me identify the circumstances in which this behavior may happen. The only pieces of modification applied to the software is the following [patch](https://github.com/flathub/org.raspberrypi.rpi-imager/blob/master/rpi-no-telemetry-by-default.patch) which disables telemetry by default.

Cheers.

maxnet, 2021-03-23:
Cherrypick 2d87a2275f9a2525f000d0f216468b9cced97e98 if you need the software to be able to run on Fedora.

ghisvail, 2021-03-23:
Thanks.

> Cherrypick 2d87a22
> if you need the software to be able to run on Fedora.

I am still left with this issue with flashing custom images on the Flatpak
version.

maxnet, 2021-03-23:
>I am still left with this issue with flashing custom images on the Flatpak version.

That is about that...
See the related issue: https://github.com/raspberrypi/rpi-imager/issues/159

ghisvail, 2021-03-23:
My bad. I could figure the relationship at first, the link to the issue
helped.

Thanks again.

closed 2021-03-26 by maxnet

ghisvail, 2021-03-26:
Thanks, this is now fixed on the Flatpak version as well.