raspberrypi/rpi-imager #186: Delete after download

m42e, 2021-03-23:
I had the issue that the imager tried to download the image, which did not happen beacause it seems some proxy settings were not respected (MacOS). 
But before downloading it already started to delete the partitions on the SD card.

At least it should check if the download is possible before deleting the SDCard content.

maxnet, 2021-03-23:
That do is a rare corner case.
Keep in mind that in the typical case of there not being Internet connectivity, Imager would not be able to download the list of OS images that can be downloaded either, so you would not be able to select an OS to download.

lurch, 2021-03-23:
If you're about to overwrite the contents of the SD card anyway (_if_ the download would have worked), what's the use-case for **not** deleting the SDCard content?

m42e, 2021-03-23:
Might be just a kind of user friendly behaviour. if it doesn't work, do not touch it at all.

commit e79467b6 (2021-11-18): Pass OS proxy settings to libcurl

Closes #186
closed 2021-11-18 by maxnet