raspberrypi/rpi-imager #188: Advanced Settings pop-up not disappearing if you change some settings (Windows)

cl-ement05, 2021-03-23:
I tried to change settings multiples times and I also tried to restart the app but the bug is still here. 

***Steps to reproduce :***
Just open the advanced settings, enable some settings (I enabled SSH, wifi and changed keyboard layout)

***Issue it self :***
You should notice that when clicking on the "save" button, the pop-up does not disappear and settings are not saved

lurch, 2021-03-23:
@maxnet Maybe it'd be worth showing a red asterisk or something to indicate that there's some fields that are still missing required info, like you get on websites if you enter your credit card number but forget to enter the expiry date? :wink:

cl-ement05, 2021-03-23:
> 
> 
> @maxnet Maybe it'd be worth showing a red asterisk or something to indicate that there's some fields that are still missing required info, like you get on websites if you enter your credit card number but forget to enter the expiry date? 😉

Well yes since this is what just happened to me lol

closed 2021-03-23 by cl-ement05

reukiodo, 2021-12-02:
Related: #318

reukiodo, 2022-10-21:
A workaround is outlined in [a forum post](https://forums.raspberrypi.com/viewtopic.php?p=2048006#p2048006):

Use RPi imager to write the SD with special options and 'password' as the password.
From the FAT32 boot drive, open the firstrun.sh in Notepad.
Find the section with cat >/etc/wpa_supplicant/wpa_supplicant.conf <<'WPAEOF'
Replace the line from psk=...gibberish... to key_mgmt=NONE
Save, close, and eject the SD card.
Put into your RPi and it after a long boot, should be on your open WiFi network!