raspberrypi/rpi-imager #191: RPi Imager downloads older bootloader version

ealap, 2021-03-25:
I was trying to flash the new bootloader version and RPi Imager indicates that it will download the latest one

> [image]

But after flashing, it turns out that it was the older `2020-09-03-vl805-000138a1` version

I further confirmed it when I checked the hash of what what I'm getting from the imager and the bootloader files I manually downloaded from the [rpi-eeprom release page](https://github.com/raspberrypi/rpi-eeprom/tags)

```
bash $ find /mnt/e/ -type f \( -not -iwholename '*System*' \) -exec sha256sum {} \+ | sort -k2,2d
bb31af9bb753d5a97ba98ecb5694475584b53210d93595b4650186da94ffddbe  /mnt/e/pieeprom.bin
abfa0cce263c684281193d749234cb6f24a313fab3403bbf1807800576df2a32  /mnt/e/pieeprom.sig
322fea1042c6c9c720ded4f1fe0ffd95dac1aa8b327e22ef57c4c0e92b9336ac  /mnt/e/README.txt
a08118fd35295fcde4911d0e0b4ea12c240dfa4ce6b451aab7957ef6d0059eae  /mnt/e/recovery.bin
6246230ecd5b472902e6a49c95e857a5e3190c4fa6c462d6a8867e9a5e523a7c  /mnt/e/vl805.bin
922792fec5e21f229d83e15a5d6e5b577b3e1f971edafa2dcdb6954e31cecf74  /mnt/e/vl805.sig
bash $
bash $ find /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2020-09-03-vl805-000138a1/ -type f -exec sha256sum {} \+ | sort -k2,2d
bb31af9bb753d5a97ba98ecb5694475584b53210d93595b4650186da94ffddbe  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2020-09-03-vl805-000138a1/pieeprom.bin
abfa0cce263c684281193d749234cb6f24a313fab3403bbf1807800576df2a32  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2020-09-03-vl805-000138a1/pieeprom.sig
322fea1042c6c9c720ded4f1fe0ffd95dac1aa8b327e22ef57c4c0e92b9336ac  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2020-09-03-vl805-000138a1/README.txt
a08118fd35295fcde4911d0e0b4ea12c240dfa4ce6b451aab7957ef6d0059eae  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2020-09-03-vl805-000138a1/recovery.bin
6246230ecd5b472902e6a49c95e857a5e3190c4fa6c462d6a8867e9a5e523a7c  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2020-09-03-vl805-000138a1/vl805.bin
922792fec5e21f229d83e15a5d6e5b577b3e1f971edafa2dcdb6954e31cecf74  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2020-09-03-vl805-000138a1/vl805.sig
bash $
bash $ find /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2021-03-18-vl805-000138a1-sd/ -type f -exec sha256sum {} \+ | sort -k 2,2d
f49b478d76465cda7da7f53b041ef16be406f510f970cb7ab0deb191f3f6b885  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2021-03-18-vl805-000138a1-sd/pieeprom.bin
6194be4882b951d0c508c8a402c7539c25146cb7610163460518c1f4f45c7222  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2021-03-18-vl805-000138a1-sd/pieeprom.sig
322fea1042c6c9c720ded4f1fe0ffd95dac1aa8b327e22ef57c4c0e92b9336ac  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2021-03-18-vl805-000138a1-sd/README.txt
1e00e443b1e3e1f912bc9ddcbfc627ba04f0c5d5eb3e22626a930e8dbb5f0ee4  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2021-03-18-vl805-000138a1-sd/recovery.bin
6246230ecd5b472902e6a49c95e857a5e3190c4fa6c462d6a8867e9a5e523a7c  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2021-03-18-vl805-000138a1-sd/vl805.bin
922792fec5e21f229d83e15a5d6e5b577b3e1f971edafa2dcdb6954e31cecf74  /mnt/c/Temp/rpi/rpi-boot-eeprom-recovery-2021-03-18-vl805-000138a1-sd/vl805.sig
```

lurch, 2021-03-25:
This is as expected, see https://downloads.raspberrypi.org/eeprom_recovery.json

I think I asked @timg236 before about this discrepancy between the "release date" (as displayed by RPi Imager) and the actual date of the EEPROM image, and he said this was by design? ping also @XECDesign as I believe it's him that updates these JSON files.

timg236, 2021-03-25:
The imager is downloading the correct version 2020-09-03. However, the "Released: 2021-03-08" string seems to refer to the "Github release" of the zip files which will change if they are repackaged.
https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2020.09.03-138a1-imager

I don't think the Github release number is useful and AFAIK we can't control it. Therefore, I think it could either be removed or replaced with a string from the JSON config file.

timg236, 2021-03-26:
@XECDesign has updated the JSON file and it now displays the correct release date

closed 2021-03-26 by ealap

XECDesign, 2021-03-29:
> I think I asked @timg236 before about this discrepancy between the "release date" (as displayed by RPi Imager) and the actual date of the EEPROM image, and he said this was by design?

As much as I like that the blame was shifted to somebody else, it was actually me. My reason for it was based on the assumption that we could potentially have a VL805 update without a bootloader update, so it made sense to me to use the github release date as the release date. However, it looks like we'd bump the bootloader version string as well in that scenario, so using the date from github only adds confusion.