raspberrypi/rpi-imager #192: Is the file extension (.ddeb) correct?

matoken, 2021-03-26:
I think the `.ddeb` in the release is a mistake for `.deb`.

```
$ file rpi-imager-dbgsym_1.6_amd64.ddeb
rpi-imager-dbgsym_1.6_amd64.ddeb: Debian binary package (format 2.0), with control.tar.xz, data compression xz
```

lurch, 2021-03-26:
Yep, looks like it https://github.com/raspberrypi/rpi-imager/releases/tag/v1.6 (as the armhf version is just `.deb`)
But I think `dbgsym` is "debug symbols", so probably not needed by 99% of people.

maxnet, 2021-03-26:
>I think the .ddeb in the release is a mistake for .deb.

The release page also has: rpi-imager_1.6_amd64.deb

You need the .ddeb in addition to that if you want to report crashes.

lurch, 2021-03-26:
@maxnet This is clearly something I know nothing about, but should `rpi-imager-dbgsym_1.6_armhf.deb` also be `.ddeb` ? :shrug:

maxnet, 2021-03-26:
>@maxnet This is clearly something I know nothing about, but should rpi-imager-dbgsym_1.6_armhf.deb also be .ddeb ? shrug

Ask Debian about the logic for the file extensions. :-)
This are the files `debbuild` spits out on each platform.

maxnet, 2021-03-26:
BTW may also be a Debian vs Ubuntu thing.
The amd64 is compiled on Ubuntu.

Either way. Closing this.

closed 2021-03-26 by maxnet