raspberrypi/rpi-imager #193: Add option to run script after image write

mkstephenson, 2021-03-27:
It would be cool if there was the option in the advanced dialog to select a script or a program to run after the image has been written. This could then be used to overwrite files in the boot drive (for example I use the cloud-init scripts in the Ubuntu Server image to configure the OS how I want it; I would want to copy these files over).

lurch, 2021-03-27:
Duplicate of #174 ?

mkstephenson, 2021-03-27:
Not that I am aware of, if I understood #174, it was a request to include a script which is run on the first/every boot of the OS which has been written to the MicroSD card.
In this case, I would like to run a script after the image has been written to the MicroSD card on the computer that is writing the image (so before the card is even inserted into the Raspberry Pi, it's a post-write step in the RPi Imager). Correct me if I'm wrong though.

lurch, 2021-03-27:
> on the computer that is writing the image

Bear in mind that RPi Imager works on Windows, MacOS and Linux - is there a cross-platform way to run a post-write script?

mkstephenson, 2021-03-27:
I believe so, from what I can understand of the library dependencies of the RPi Imager it uses Qt 5 as a framework. A cursory glance over the docs would suggest that [QProcess](https://doc.qt.io/qt-5/qprocess.html) would allow you to launch an external program with optional parameters.

maxnet, 2021-03-29:
I do not know if this a feature we should have.
There may be other more suitable solutions than Imager if you want to write a lot of images, and need post-installation scripts with that.

As for the technicalities of running a script in a cross-platform way, do be aware that while QProcess can indeed start a program on every platform, however under what privileges this program will run does differ in the case of Imager.
Imager runs as a normal user on Linux and Mac OS X.
And privileged under Windows.
With the side effect that any script you will start with QProcess will have Administrator access on Windows as well.
This may allow privilege escalation attacks under certain circumstances.

CharlesGodwin, 2021-03-31:
My solution to this is to build the image with the authentication key so my first ssh connection is simple.

I use windows but it should be similar on other systems.
I just do this. I build a script of what I want to do and run it on the pi

`ssh pi@hostname  <myscript.sh`

If I need files on the pi `scp filename pi@hostname`

I have some cookbook type scripts that I always run to do setup for things like docker, email transfer agents etc.

tdewey-rpi, 2024-02-06:
Rejecting this feature request for the reasons provided by @maxnet above.

closed 2024-02-06 by tdewey-rpi