raspberrypi/rpi-imager #194: [OS Request] SimplyPi

AlbertMN, 2021-03-28:
Hi there :)

I'm the co-creator of [SimplyPrint](https://simplyprint.io/) which has the [SimplyPi](https://github.com/SimplyPrint/SimplyPi) distro. It's a fork of [OctoPi](https://github.com/raspberrypi/rpi-imager/issues/162) which was added to the installer a few weeks ago, that comes with the SimplyPrint plugin installed which allows the user to control their 3D printer from anywhere (and a few other small tweaks).

I too have just copied what I saw in #83 as well as #162, and linking to a 40x40 image.

Note that it should go in the `Other spesific purpose OS` section.
```
{
    "os_list": [
        {
            "name": "SimplyPi",
            "description": "Bring your 3D printer(s) with you anywhere with SimplyPrint on top of OctoPrint.",
            "icon": "http://cdn.simplyprint.io/i/static/rpi-imager/icon_gradient.png",
            "subitems_url": "https://simplyprint.io/rpi-imager/rpi-imager-simplypi.json"
        }
    ]
}
```
And the json file is in place:
https://simplyprint.io/rpi-imager/rpi-imager-simplypi.json

The SimplyPi GitHub repository is not that "active" or has many stars, as it's usually always downloaded via this download page; https://simplyprint.io/downloads, from where it has over the past 2 months have gotten 2762 downloads, with between 500-700 people using the distro every day, gaining about 10-30 downloads per day, rapidly growing, so it'll be downloaded frequently.

Loving the RPi imager! :) Thanks for making it.

ghollingworth, 2022-02-02:
Can you email me at gordon@ this organisation dot com

AlbertMN, 2022-02-02:
Email sent @ghollingworth - thanks :)

ghollingworth, 2022-02-02:
Added

closed 2022-02-02 by ghollingworth