raspberrypi/rpi-imager #196: More welcoming changelog for

claremacrae, 2021-03-29:
https://github.com/raspberrypi/rpi-imager/releases/tag/1.6.1 Says:

> Advanced options: if user configured **he** wants to use SSH, but forget to enter password, highlight "password" label red to draw attention to that, instead of just refusing to save settings.

would you please consider changing **he** to **they** in the text of the release note and any copies of it, to make the message more inclusive?

claremacrae, 2021-03-29:
Crikey, that was quick!

maxnet, 2021-03-29:
Changed it.

Did not had the intent to claim any group of people is more likely to forget to complete required fields.
Just struggling with writing proper English sentences sometimes (secondary language for me).

closed 2021-03-29 by maxnet

claremacrae, 2021-03-29:
I was more thinking girls and women might think the tool was not for them.

But of course, I should have said I realised that this wasn’t the intention - and thank you for this fabulous tool!