raspberrypi/rpi-imager #200: RPi Imager 1.6.1 can't be built on Ubuntu 18.04

lurch, 2021-03-31:
I followed the instructions in the [README](https://github.com/raspberrypi/rpi-imager/tree/1.6.1#debianubuntu-linux) but it fails with a build error:
```
/home/andrew/github/raspberrypi/rpi-imager/downloadthread.cpp: In member function ‘bool DownloadThread::_customizeImage()’:
/home/andrew/github/raspberrypi/rpi-imager/downloadthread.cpp:935:45: error: ‘class QString’ has no member named ‘back’
             if (!folder.isEmpty() && folder.back() == '\\')
                                             ^~~~
CMakeFiles/rpi-imager.dir/build.make:254: recipe for target 'CMakeFiles/rpi-imager.dir/downloadthread.cpp.o' failed
make[3]: *** [CMakeFiles/rpi-imager.dir/downloadthread.cpp.o] Error 1
```
Looks like [QString.back()](https://doc.qt.io/qt-5/qstring.html#back) was added in Qt 5.10, but Ubuntu 18.04 only includes Qt 5.9.5

maxnet, 2021-03-31:
>Looks like QString.back() was added in Qt 5.10, but Ubuntu 18.04 only includes Qt 5.9.5

Just eeplace it with .right(1) for now.

Not sure if we can keep supporting 5.9.5 in the long run.
Some changes needed to move to Qt 6 do not work in 5.9.5 Somewhat related issue: https://github.com/raspberrypi/rpi-imager/pull/134

lurch, 2021-03-31:
Ah, before seeing your reply I just fixed it with:
```patch
diff --git a/downloadthread.cpp b/downloadthread.cpp
index 1fa0e5c..010aa54 100644
--- a/downloadthread.cpp
+++ b/downloadthread.cpp
@@ -932,7 +932,7 @@ bool DownloadThread::_customizeImage()
         for (auto mp : mountpoints)
         {
             folder = QString::fromStdString(mp);
-            if (!folder.isEmpty() && folder.back() == '\\')
+            if (!folder.isEmpty() && folder.at(folder.size() - 1) == '\\')
                 folder.chop(1);
             configFilename = folder+"/config.txt";
 
```

lurch, 2021-03-31:
...but it also builds with the slightly simpler:
```patch
diff --git a/downloadthread.cpp b/downloadthread.cpp
index 1fa0e5c..fe04290 100644
--- a/downloadthread.cpp
+++ b/downloadthread.cpp
@@ -932,7 +932,7 @@ bool DownloadThread::_customizeImage()
         for (auto mp : mountpoints)
         {
             folder = QString::fromStdString(mp);
-            if (!folder.isEmpty() && folder.back() == '\\')
+            if (!folder.isEmpty() && folder.right(1) == '\\')
                 folder.chop(1);
             configFilename = folder+"/config.txt";
 
```
:smile:

commit 0c6a9359 (2021-04-05): Avoid using QString::back() for compatibility with older Linux distros

Old Qt versions lack QString::back()
Closes #200
closed 2021-04-05 by maxnet