raspberrypi/rpi-imager #202: RPi Imager 1.6.1 keep crashing on MacOS Big Sur 11.2.3

GaoShin, 2021-04-04:
After upgrade to version 1.6.1 it keeps crashing when open.
The crash log is below

`Process:               rpi-imager [87114]
Path:                  /Applications/Raspberry Pi Imager.app/Contents/MacOS/rpi-imager
Identifier:            org.raspberrypi.imagingutility
Version:               1.6.1 (1.6.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           rpi-imager [87114]
User ID:               501

Date/Time:             2021-04-04 13:09:06.972 +0800
OS Version:            macOS 11.2.3 (20D91)
Report Version:        12
Bridge OS Version:     5.2 (18P4347)
Anonymous UUID:        6E53377D-BBF9-2BB9-DD69-1CD8E7CD2F4E

Sleep/Wake UUID:       40788867-A15A-4DE4-991F-9DE3169109A6

Time Awake Since Boot: 96000 seconds
Time Since Wake:       1000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000020
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [87114]

VM Regions Near 0x20:
--> 
    __TEXT                      1031d1000-103329000    [ 1376K] r-x/r-x SM=COW  /Applications/Raspberry Pi Imager.app/Contents/MacOS/rpi-imager

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	0x00007fff203379bf objc_release + 31
1   com.apple.CoreFoundation      	0x00007fff20532761 __RELEASE_OBJECTS_IN_THE_ARRAY__ + 118
2   com.apple.CoreFoundation      	0x00007fff205326a4 -[__NSArrayM dealloc] + 279
3   org.raspberrypi.imagingutility	0x00000001031ddf88 main + 3608
4   libdyld.dylib                 	0x00007fff204b4621 start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff20495458 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff20495458 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff20495458 start_wqthread + 0

Thread 4:: QQmlThread
0   libsystem_kernel.dylib        	0x00007fff2046b4fe poll + 10
1   org.qt-project.QtCore         	0x0000000104c15830 qt_safe_poll(pollfd*, unsigned int, timespec const*) + 608
2   org.qt-project.QtCore         	0x0000000104c17021 QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 833
3   org.qt-project.QtCore         	0x0000000104bb9b6f QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
4   org.qt-project.QtCore         	0x00000001049e9c43 QThread::exec() + 131
5   org.qt-project.QtQml          	0x00000001040fd269 0x103e86000 + 2585193
6   org.qt-project.QtCore         	0x00000001049eaba9 0x1049c8000 + 142249
7   libsystem_pthread.dylib       	0x00007fff20499950 _pthread_start + 224
8   libsystem_pthread.dylib       	0x00007fff2049547b thread_start + 15

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000010
  rdi: 0x00006000015b6600  rsi: 0x0000000000000001  rbp: 0x00007ffeeca2e7d0  rsp: 0x00007ffeeca2e798
   r8: 0x00000000000000cb   r9: 0x00000000000007fb  r10: 0x00007fff884f3670  r11: 0x00007fff2053258d
[... 364 lines cut ...]
    0x7fff6ca44000 -     0x7fff6ca80fff  libGLU.dylib (18.1.1) <F0DD35B5-3692-3BE5-AD3D-4F6B237EF6AD> /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGLU.dylib
    0x7fff6cc16000 -     0x7fff6cc20fff  libGL.dylib (18.1.1) <68ABAADE-2DB4-3412-9F36-CA1AEAC8E9F6> /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGL.dylib
    0x7fff6e057000 -     0x7fff6e0affff  com.apple.opencl (4.5 - 4.5) <40B70A3B-D981-3E4A-9BEC-F22EF84118F5> /System/Library/Frameworks/OpenCL.framework/Versions/A/OpenCL
    0x7fff6f12e000 -     0x7fff6f135fff  com.apple.agl (3.3.4 - AGL-3.3.3) <25240534-44D6-3350-9C56-0D0CEAC3830B> /System/Library/Frameworks/AGL.framework/Versions/A/AGL
    0x7fff7821c000 -     0x7fff78223fff  libRosetta.dylib (203.30) <44C477D9-013F-3A6D-A9FE-68A89214E6A5> /usr/lib/libRosetta.dylib

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 260945
    thread_create: 0
    thread_set_state: 0

VM Region Summary:
ReadOnly portion of Libraries: Total=727.9M resident=0K(0%) swapped_out_or_unallocated=727.9M(100%)
Writable regions: Total=623.7M written=0K(0%) resident=0K(0%) swapped_out=0K(0%) unallocated=623.7M(100%)
 
`

maxnet, 2021-04-04:
If it work in 1.6 (without .1) I am guessing the new code to detect the preferred language the program should be displayed in is not working properly on your system.
Unfortunately, I am not able to reproduce the issue on the single Mac Mini I have. (Nor was I am able to reproduce the problem the new code was trying to solve, which make solving issues like this hard).

If you go the Mac OS settings, and go to "Language & Region", do you see any abnormalities in that screen?
There do is at least one language configured in the "Preferred languages" box to the left?

maxnet, 2021-04-04:
Can you try if this build works any better on your system?

[Imager-20210404-mac.zip](https://github.com/raspberrypi/rpi-imager/files/6255260/Imager-20210404-mac.zip)

GaoShin, 2021-04-05:
Tried the old release version 1.6 ,it did work on my system.
Also tried your build version (1.6.2?)and it's ok,no crash.The program show with my preferred language.

> If you go the Mac OS settings, and go to "Language & Region", do you see any abnormalities in that screen?
There do is at least one language configured in the "Preferred languages" box to the left?
There seems no abnormal with "Preferred languages" setting

maxnet, 2021-04-05:
>Also tried your build version (1.6.2?)and it's ok,no crash

Thanks for the info.

Was indeed a very obvious memory management bug there. (but it showed no symptoms on my own system, so had been overlooked)

maxnet, 2021-05-26:
Fixed in 1.6.2

closed 2021-05-26 by maxnet