raspberrypi/rpi-imager #204: Extra confirmation for writing an image on external HDD instead of SD or USB memory

epsi1on, 2021-04-08:
# Preface 

I wanted to write RPI image to my SD card, puted SD card into slot but my computer recognized the SD card with 10 sec delay and by mistake I did chose the only option, a 2TB external HDD and Boom, hdd is blank now! :( 

[image]

It hurts ...

This happened to me and maybe other users, so I think it would be good to have this check before start wiping disk:

# Actual Feature Request

If user select an external HDD instead of SD card then app shows another confirmation message, with something saying user that this is a HDD not a memory card you are writing on!!! are you certainly sure??

There is such security mechanisms in other software like GitHub itself. When you want to delete a repository, then github ask you to write repo name in order to confirm the delete operation:

[image]

Detecting external HDD and memory cards TF can be based on size of storage, for example a 500GB TF card is rarely used by users so extra confirmation on such sizes would not make serious problems I think...

[image]

maxnet, 2021-04-08:
The problem is that there is no reliable way to detect type of storage.
Size may work if the hard disk is sufficiently large. 
However what about a 128 GB external SSD versus a 128 GB SD card?
No reliable way to tell those apart.

lurch, 2021-04-08:
In Etcher there's a check if a drive is "unusually large", which IIRC was originally 64GB but is now 128GB

https://github.com/balena-io/etcher/blob/master/lib/gui/app/components/drive-status-warning-modal/drive-status-warning-modal.tsx#L20
https://github.com/balena-io/etcher/blob/master/lib/shared/drive-constraints.ts#L145

But I concur with @maxnet that for RPi Imager there's no way to tell whether a USB Mass-Storage-Device is a SD card, an SSD, a HDD or a USB flash drive. _Some_ SD card-readers provide this info, but many of them just present the SD card to the OS as a "generic USB drive".

tdewey-rpi, 2024-02-06:
Rejecting this feature request:

1. Large device formatting is not an unexpected flow. It's entirely conceivable a user might choose to use Raspberry Pi Imager to install to a large disk for use in a NAS, etc.
2. With no reliable mechanism to determine the type of disk, and using size unviable through point (1), this feature cannot be implemented in a satisfactory manner.

closed 2024-02-06 by tdewey-rpi